MarlinFirmware/Configurations #643: [Issue] Display malfunction when multiple temperature sensors are used

RE-RaytreE, 2021-12-29:
Broad: FlyF407ZG
Display: MKS MINI12864 V3.0
I was thinking building a new printer with FlyF407ZG broad, which has 8 temperature sensor plugs on-board. But when I define more than 5 sensors, the LCD can't show all the readings at status screen. 
[image]

I was thinking display them by some numbers rather than images. Or scroll the readings every certain time.
I'm struggled to solve this problem, what can I do?
BTW, I modified the Marlin 2.0.9.1 Firmware in order to use MKS Mini12864 V3 LCD display, which is the only spare LCD I have...The firmware I use is provided as below.
[ConfigFiles.zip](https://github.com/MarlinFirmware/Configurations/files/7788691/ConfigFiles.zip)

ellensp, 2021-12-29:
lcd is only 128 pixels across.  
8 hotends is a big ask....
16 pixels per extruder
4 chars each tempature, thats is a 3pixel font (+1 pixel inter char spacer)

RE-RaytreE, 2021-12-29:
> How about just numbers？Shows the sensor Type and readings Only.

T1 20°    T4  210°  TB 60°
T2 19°    T5  245°  TC 40°
T3 190°  T6  200° 

That way may save some pixels?

descipher, 2021-12-29:
Nope. But you could try to workout a left to right scrolling function.

thisiskeithb, 2021-12-29:
At one point, thinkyhead discussed possibly swapping out extruders on each screen update or disabling graphics for inactive hotends.

Also, this isn’t a bug/feature request for the configuration repo. Please open a Feature Request in the [main Marlin repo](https://github.com/MarlinFirmware/Marlin/issues/new?assignees=&labels=T%3A+Feature+Request&template=feature_request.yml&title=%5BFR%5D+%28feature+summary%29).

closed 2021-12-29 by thisiskeithb