MarlinFirmware/Configurations #644: [FR] Ender 3 Pro with SKR mini E3 V3.0

HoreaM, 2021-12-29:
### Description

I searched in the configurations folder for the Ender 3 Pro, but it seems like the only available configuration files are for the SKR mini E3 v1/1.2/2 boards. It would be of great help if we could get an "official" configuration file for the Ender 3 Pro with the new SKR mini E3 V3.0 board, as I'm not entirely sure what needs to be adjusted from the previous SKR mini E3 versions. Thank you!

### Configuration Details

* Manufacturer: Creality
* Model: Ender 3 Pro
* Motherboard: SKR mini E3 V3.0

HoreaM, 2022-01-15:
I just noticed that the Ender 3 has a configuration file for the SKR mini E3 V3.0. Could I use that one for my Ender 3 Pro? Thank you!

thisiskeithb, 2022-01-15:
> Could I use that one for my Ender 3 Pro?

Yes. As far as Marlin is concerned, the only difference is the printer name and status icon text.

thisiskeithb, 2022-01-15:
Having said that, here's a config: [Ender-3 Pro - BigTreeTech SKR Mini E3 3.0.zip](https://github.com/MarlinFirmware/Configurations/files/7875628/Ender-3.Pro.-.BigTreeTech.SKR.Mini.E3.3.0.zip)

I'll put in a PR later.

thisiskeithb, 2022-01-15:
Added in https://github.com/MarlinFirmware/Configurations/pull/660

closed 2022-01-15 by thisiskeithb