MarlinFirmware/Configurations #645: [FR] Support for Creality board v24S1 v301 for Ender 3 S1

leofaoro, 2021-12-30:
Hello
I’d love to install a custom Marlin FW on my new Ender 3. A few annoying things have been disabled by default. 
Anyway the board is new, powered by a STM32F103RCT6. I tried to trace the pins but couldn’t do it properly. 

Would greatly appreciate an official config. 
thanks

ellensp, 2021-12-30:
[image]
[image]

leofaoro, 2021-12-30:
Yeah that's my post on Reddit :) Here are a couple more that are more clear, also with the back of the board.

[image]
[image]

leofaoro, 2021-12-31:
HEATER BED is on pin 63 (VSS_3). Also for the CR-Touch,  pin 2 (PC13) = servo pin, pin 3 (PC14) = probe pin. Updated Peter's spreadsheet.

ellensp, 2021-12-31:
"HEATER BED is on pin 63 (VSS_3)" 
That is a gnd pin.  That cannot be correct 

The normal arrangement for a heated bed is the +v is permanently connected but the -v is connected via a mosfet
When the bed is off both bed pins are at a positive voltage so no current flows. When the  bed is turned on the -ve leg is connected to gnd allowing current to flow.

leofaoro, 2022-01-01:
> "HEATER BED is on pin 63 (VSS_3)" That is a gnd pin. That cannot be correct
>
> The normal arrangement for a heated bed is the +v is permanently connected but the -v is connected via a mosfet When the [...]

Oh no, I was tracing the wrong leg then. It went from the STM32 to a resistor, to some logic level chip, then out to the mosfet, then to the bed....

leofaoro, 2022-01-03:
Ok I have the printer sort of working. Issues:
- The extruder fan does not turn on. 
- the part cooling fan makes a strange sound, not sure if it's PWM related. It works and is controllable
- the screen does not work

Here's my commit on the my 2.0.9.3 fork (not sure I'm doing this the right way, first time contributing) 
https://github.com/leofaoro/Marlin/commits/2.0.x

ellensp, 2022-01-03:
Documenting from discord chat. 
The screen does not seem to be the same as a e3v2 screen

[image]

From the updated source provided from creality 

[Ender-3s1-hw24s1_301-sw-V1.0.5_C-CRTouch.zip](https://github.com/MarlinFirmware/Configurations/files/7800175/Ender-3s1-hw24s1_301-sw-V1.0.5_C-CRTouch.zip)

It has a dwin_set folder, with files for the e3v2 screen, but it also has a private folder which contains the same information as the dwin_set but in a different format.

Current thinking is that this new lcd uses the private directory for its data.
This screen is not compatible with Marlin at this time as marlin expects the Marlin DWIN_SET

leofaoro, 2022-01-03:
Some progress. Indeed the display will update only if provided a "private" folder. I tried the DWIN_SET from the E3V2 config example in Marlin, renamed it to "private" it took but only blank screen. Also tried the "private" folder provided in Creality's latest firmware update (1.0.5) above, and constantly shows the splash page "Creality". 

Also confirmed pinout is exactly the same for both LCD screens.

leofaoro, 2022-01-03:
Extruder fan identified and added to pins file + config adv. Working.
https://github.com/leofaoro/Marlin/commit/5e8f206a15270e267340702035c9c9a2b4edb69e

Geo-Warrior, 2022-01-06:
Hi. I'm probably better suited as a lurker here as a lot of this is over my head. If there's anything I can help with, just throw it out there and I'll do my best. Thanks for the work you're doing.

skilly00, 2022-01-18:
What is left to do on this project?  Leo has it been finished or do you need help with anything?  I have this printer and would be happy to help id anything you still have outstanding.

leofaoro, 2022-01-18:
> What is left to do on this project? Leo has it been finished or do you need help with anything? I have this printer and would be happy to help id anything you still have outstanding.

Hey! The screen is still not working. It’s totally undocumented and I don’t have the skills to reverse engineer it. Everything else seems to be working.

skilly00, 2022-01-18:
Ok cool.  well maybe I'm missing something but can't you just include the Firmware from the official files for the screen?

leofaoro, 2022-01-18:
> Ok cool. well maybe I'm missing something but can't you just include the Firmware from the official files for the screen?

Nope. Doesn't work.

mriscoc, 2022-01-18:
> > Ok cool. well maybe I'm missing something but can't you just include the Firmware from the official files for the screen?
> 
> Nope. Doesn't work.

I don't have a Ender3S1 but have some interest in try to develop something for that printer. We need to know exactly what mean "Doesn't work", the screen is permanent in black, the display shows boot screen and then black, the screen shows a bad user interface, etc.

leofaoro, 2022-01-18:
Fair. Currently the screen turns on and stays on the boot screen. The button when pressed does beep. That’s about all it does. If someone has an ender 3 v2 screen and wants to try that could be helpful.

mriscoc, 2022-01-18:
OK, if the boot background is there, then that only ensures that the display is correctly energized and reset. Could be interesting to know if the printer works fine using Octoprint and last Marlin bugfix, for example.

leofaoro, 2022-01-18:
> OK, if the boot background is there, then that only ensures that the display is correctly energized and reset. Could be interesting to know if the printer works fine using Octoprint and last Marlin bugfix, for example.

Yes. It works using my pins file and connects/prints with Octoprint. As I said, the only thing that is not working is the screen.

mriscoc, 2022-01-18:
> Yes. It works using my pins file and connects/prints with Octoprint. As I said, the only thing that is not working is the screen.

Then the next thing is to identify is the serial link between the mainboard and the display.

leofaoro, 2022-01-18:
> Then the next thing is to identify is the serial link between the mainboard and the display.

Yes, and that goes beyond my skills :)

mriscoc, 2022-01-18:
I need a S1 here, :-D

skilly00, 2022-01-18:
Looks like the pins are labeled in the above picture from  ellensp.

mriscoc, 2022-01-19:
> How would one go about identifying the serial link? I have the display and am happy to help if you can point me to some kind of guide. I have a CC Debugger would that work?

I don't know if the correct serial interface in the board was enabled in the @leofaoro files.   A CC Debugger could help. I'm assuming that the firmware compilated is issuing some commands to the display at the start up, that serial activity could be noted by the debugging device if the firmware is using the correct pin.

skilly00, 2022-01-19:
Ok, @leofaoro said above that it was using the same pins.  They are labeled on the screenshot provided by @ellensp but I can't find a screenshot from the V2 to confirm they are the same.   So to be clear I should be hooking up to the 10 pin from the printer at boot to see if its getting the commands your thinking are sent.

ellensp, 2022-01-19:
It is the same
[image]

mriscoc, 2022-01-19:
The problem is to know if the firmware is using the correct serial interface and pin enables of the display interface in the S1 SoC

leofaoro, 2022-01-19:
I went ahead and mapped the pins to the LCD ribbon cable. See below. 

DOUT = PA3
DIN = PA2
BEEP = PB13
ENT = PB14
A = PB15
B = PB12

I don't know what these mean in terms of mapping to Marlin pins. I know there's BTN_ENC, BTN_EN1, BTN_EN2, BEEPER_PIN listed for DWIN_E3V2. 

[image]

leofaoro, 2022-01-19:
> The problem is to know if the firmware is using the correct serial port and pin enables of the display interface in the S1 SoC. @leofaoro, the display beeps when the enconder is pressed? The display beeps when you send a M300?

the display beeps when buttons is pressed yes. Haven't tested M300.

ellensp, 2022-01-19:
PA3/PA3 is USART2_TX/USART2_RX  ie uart port 2

mriscoc, 2022-01-19:
> PA3/PA3 is USART2_TX/USART2_RX ie uart port 2

That is an advance, in the Ender3v2 the Serial port is 3:
```cpp
#define LCD_SERIAL_PORT 3
```
Then, @leofaoro change that to 2 in `Conditionals_LCD.h` and test.

skilly00, 2022-01-19:
not sure if this helps at all but I just put the display from the s1 into a v2 and it seemed to work just fine.  I went through all the menus and had full functionality.

mriscoc, 2022-01-19:
> 

T

> not sure if this helps at all but I just put the display from the s1 into a v2 and it seemed to work just fine. I went through all the menus and had full functionality.

That helps! then the serial speed, protocol and icon assets are the same!

ellensp, 2022-01-19:
As I already said its is a dwin clone, using dwin protocol.
but asset binary are in a different format.
It uses assets from the private directory, not the dwin_set
see example in https://github.com/MarlinFirmware/Configurations/files/7800175/Ender-3s1-hw24s1_301-sw-V1.0.5_C-CRTouch.zip

mriscoc, 2022-01-19:
But the test from @skilly00 confirm that, I didn't have a confirmation before or at less I didn't see it.

ellensp, 2022-01-19:
Its clear to me, as the official update doesn't have two sets of controller firmware only two sets of assets. So bot screens must be using same protocol

ellensp, 2022-01-19:
I think efforts should be directed at finding how to covert dwin_set files to private files

mriscoc, 2022-01-19:
Ok, please test with this repo: https://github.com/mriscoc/Marlin_Ender3S1
Precompiles binary here: https://github.com/mriscoc/Marlin_Ender3S1/releases/latest
Caution this is a Beta (or Alpha) version.

leofaoro, 2022-01-19:
> Ok, please test with this repo: https://github.com/mriscoc/Marlin_Ender3S1
> Precompiled binary here: https://github.com/mriscoc/Marlin_Ender3S1/releases/latest
> Please check if the display is working now. Caution this is a Beta (or Alpha) version.

Screen works!!  Haven’t tested anything else but assuming you use my initial files, should be all good. Will test more later.

skilly00, 2022-01-19:
Great news!  Thank you @mriscoc @leofaoro and @ellensp for all the work done  I'm finishing a print on mine around 2PM EST and will load it as well to test.

mriscoc, 2022-01-19:
@leofaoro did you test the changes to see if the display works?

mriscoc, 2022-01-19:
Excellent!

skilly00, 2022-01-20:
Sorry a bit later than I said but I can confirm it is working as expected with your recompiled version.  I did a few prints and bed leveling, local and with OctoPrint and all has come out fine!

mriscoc, 2022-01-20:
> Sorry a bit later than I said but I can confirm it is working as expected with your recompiled version. I did a few prints and bed leveling, local and with OctoPrint and all has come out fine!

What about the BLTouch and filament runout sensor?

skilly00, 2022-01-20:
It has a CR touch but yes that worked with the auto leveling on the screen and auto Z offset as well as in OctoPrint with Unified Gcode in bedlevelvisualizer plugin.  I didn't test the runout sensor but can do that now.

mriscoc, 2022-01-20:
> It has a CR touch but yes that worked with the auto leveling on the screen and auto Z offset as well as in OctoPrint with Unified Gcode in bedlevelvisualizer plugin. I didn't test the runout sensor but can do that now.

There is an end-stop diagnostics page in the Advance/Filament Settings menu

skilly00, 2022-01-20:
Yeah then it is working correctly.  I saw that when running through the menu after loading the first time.  I can also say it says Open when the filament is removed.

mriscoc, 2022-01-20:
> Yeah then it is working correctly. I saw that when running through the menu after loading the first time. I can also say it says Open when the filament is removed.

Thank you for your report!

Wassabichan, 2022-01-20:
i have testet it and recognized the my raspberry isn`t working with the firmware
is this normal and is there a way to go back to stock firmware?

Wassabichan, 2022-01-20:
i did, and then i wannt to report that my screen was only blue and black and the screen output  wasn`t working proplably.
it isn't like i want to say it`s trash  i just wanna give feedback and i wannja help were i can

skilly00, 2022-01-20:
@mriscoc I didn't report it because I figured I needed to change my start gcode most likely but when I use Octoprint, after homing/heating and leveling the printer hangs for 5 or so min before it starts.  This could be something in my start code I haven't dug into but if @Wassabichan is having issues with OctoPrint it may be related to the Firmware.

Wassabichan, 2022-01-20:
ând is it normal thet the interface is only blue with dark writing? or is only mine like that?

mriscoc, 2022-01-20:
> @mriscoc I didn't report it because I figured I needed to change my start gcode most likely but when I use Octoprint, after homing/heating and leveling the printer hangs for 5 or so min before it [...]

Please confirm that is not the warm-up time of the bed/nozzle

mriscoc, 2022-01-20:
> ând is it normal thet the interface is only blue with dark writing? or is only mine like that?

Please post an image.

Wassabichan, 2022-01-20:
sry but i can`t because i already have the normal firmware on it again

Wassabichan, 2022-01-20:
but i can do it tomorrow, if it can help

Sebazzz, 2022-01-20:
@mriscoc I suggest the following diff to your config:

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 091d7e134f..208447043b 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
 /**
@@ -123,7 +123,7 @@
  *
  * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
  */
-#define BAUDRATE 250000  // MRiscoC increase serial performace
+#define BAUDRATE 115200  // MRiscoC increase serial performace
 #define BAUD_RATE_GCODE     // Enable G-code M575 to set the baud rate  // MRiscoC Enables change the baudrate
 
 /**
@@ -1198,7 +1198,7 @@
  *     |    [-]    |
  *     O-- FRONT --+
  */
-#define NOZZLE_TO_PROBE_OFFSET { -37, -39, -3 }  // Ender3S1 Configs
+#define NOZZLE_TO_PROBE_OFFSET { -37, -39, -3.25 }  // Ender3S1 Configs
 
 // Most probes should stay away from the edges of the bed, but
 // with NOZZLE_AS_PROBE this can be negative for a wider probing area.
@@ -1401,8 +1401,8 @@
 #define Y_BED_SIZE 230  // MRiscoC Max usable bed size
 
 // Travel limits (mm) after homing, corresponding to endstop positions.
-#define X_MIN_POS 0
-#define Y_MIN_POS 0
+#define X_MIN_POS -5
+#define Y_MIN_POS -5
 #define Z_MIN_POS 0
 #define X_MAX_POS 248  // MRiscoC Stock physical limit
 #define Y_MAX_POS 231  // MRiscoC Stock physical limit
@@ -1574,7 +1574,7 @@
  * leveling immediately after G28.
  */
 //#define RESTORE_LEVELING_AFTER_G28
-//#define ENABLE_LEVELING_AFTER_G28
+#define ENABLE_LEVELING_AFTER_G28
 
 /**
  * Auto-leveling needs preheating
```

- Baudrate: My device didn't support baudrate 250000, besides it is unnecessarily high (unless you really need to transfer files over serial)
- `ENABLE_LEVELING_AFTER_G28`: Always enable bed leveling mesh
- X_MIN_POS / Y_MIN_POS: Otherwise X=0,Y=0 were outside the bed

May I request the source of your binary build btw, as per GPL?

leofaoro, 2022-01-20:
Everything seems to be working for me. 
The only minor issue is the fonts seem to have a different background so there are a few pixels around the font where the antialiasing isn't clear. Obviously minor.

Wassabichan, 2022-01-20:
i`m gonna take pictures tomorrow, but i had the problem that there weren't any pictures on the screen, just blank boxes and i couldn't see on what box i am

Wassabichan, 2022-01-20:
i could do them right now but i don't know how to get back the M503 Data on the printer

mriscoc, 2022-01-20:
@Sebazzz the source is in the S1 repo

[64 more replies]

closed 2022-01-26 by thinkyhead