MarlinFirmware/Configurations #647: [FR] (configuration request title)

chapito, 2021-12-31:
### Description

### Configuration Details
trying to to configure a Ender 3 Pro for new SKR BTT E3 Turbo board. there's no listing in the config folder (only for Ender3 and V2?) When I use those configs I get a VS code compile error (unable to compile)?

* Manufacturer: Creality
* Model: Ender 3Pro
* Motherboard: BTT SKR E3 Turbo

#### Additional Information

* Provide links to the requested printer/configuration.

ellensp, 2021-12-31:
there are 7 example configuration files for the Ender 3 pro
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20Pro
The closest fit being BigTreeTech SKR 1.4 Turbo
If you use that config and update motherboard with
  #define MOTHERBOARD BOARD_BTT_SKR_E3_TURBO

It compiles without issues

ellensp, 2021-12-31:
But take not of the readme. this confis is for the following hardware configuration
https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.0.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%201.4%20Turbo/README.md

thisiskeithb, 2021-12-31:
Or use these: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20E3%20Turbo

thisiskeithb, 2021-12-31:
The only difference (as far as Marlin is concerned) between the Pro and standard Ender-3 is the status screen icon and printer name.

ellensp, 2021-12-31:
Please update title to reflect feature request

thinkyhead, 2022-01-26:
Duplicate of #649

closed 2022-01-26 by thinkyhead