MarlinFirmware/Configurations #649: [FR] Configurations for all 3D printers and every possible permutation

thinkyhead, 2022-01-05:
This is a catch-all for all "me too" configuration requests.

skilly00, 2022-01-18:
Anyone working on the Ender 3 S1?  Lots of changes over earlier models and it has a new Board.

ColinM9991, 2022-01-27:
This mightn't help at all but I figure it's worth sharing.

Several months back I took a cut of Jyer's Marlin fork and added a workflow to compile firmware for a SKR Mini E3.

Most changes are in the [following commit](https://github.com/ColinM9991/Marlin/commit/a48937d426eb5592050a2bf4407acc6262c81e16#diff-87ee5504a3e25ac558b343724c905f2f7949e8cec3d92b9c4300bb922afa164f)

Individual tweaks are located in two additional commits on the [following branch](https://github.com/ColinM9991/Marlin/tree/skr-mini-e3)

JeroenR123, 2022-01-29:
> Anyone working on the Ender 3 S1? Lots of changes over earlier models and it has a new Board.

Metoo

pichoncin37, 2022-04-27:
buenas , necesitaria ayuda para configurar la bricolemon lite en mi ender 3, no me imprime en el centro esta descuadrada la impresion, y el eje x intenta seguir hacia la izquierda aun estando el endstop

walterp1982, 2022-05-04:
I'd like to request configuration for the Mingda Magician X with the (S) board. Please let me know if I can be of service.

UKIkarus, 2022-05-21:
> I'd like to request configuration for the Mingda Magician X with the (S) board. Please let me know if I can be of service.

Me too, I have this printer and I have the pinout provided by Mingda, if anyone can lend a hand I'll gladly grab any info I can from them to assist.

[BC.DZ.PC000007_PIN_info.txt](https://github.com/MarlinFirmware/Configurations/files/8748107/BC.DZ.PC000007_PIN_info.txt)
[tft35.txt](https://github.com/MarlinFirmware/Configurations/files/8748108/tft35.txt)

Hope these help.

duchengyao, 2022-06-11:
Any plan to add SermoonV1 support?

https://3dprintbeginner.com/creality-sermoon-v1-pro-review/

ellensp, 2022-06-11:
@duchengyao Yet another undocumented motherboard form Creality using a STM32F401RC, no source code provided even though it a requirement to use Marlin. So no pins files...  Nothing to work from.
If you have brought one, demand the firmware source.   
Once you have that we can consider it

duchengyao, 2022-06-11:
> @duchengyao Yet another undocumented motherboard form Creality using a STM32F401RC, no source code provided even though it a requirement to use Marlin. So no pins files... Nothing to work from. If you have brought one, demand the firmware source. Once you have that we can consider it

@ellensp  I have tried to demand it, but the official firmware source code is not provided. Is there a way to modify the config file empirically, I can try it.

XForce6453, 2022-06-11:
Marlin (bugfix 2.1.x) Config Files for Vellemann K8200 with DirectDrive Extruder, Z-Axis-Upgrade and RepRap_Full_Smart_Grapic_Display.
Everything fully functional incl. SD-Card function (No autodetect of SD / Have to be manualy initialized via menu).
The hidden Game-Menu is also preactivated.
Note: The file "menu_main.cpp" have to be replaced too due to a change for continous visibility of the SD-Card Init. menu even there is no SD-Card in Slot so you can initialize an SD-Card without the need of autodetection.

UKIkarus, 2022-06-15:
I now have the sourcecode for Mingda Magician X.
Will follow up after work, hopefully this will be of use for other devices using the same board?

kukumagi, 2022-06-17:
Hello. Has anyone updated Simax Mi-M200 firmware? What config should I use?

bmccane, 2022-06-20:
Are there plans to create a configuration for TronXY X5SA with the CXY-V6-191121 board?
It has STM32F446ZET

sgparry, 2022-06-27:
Please add up-to-date and working configuration for Vertex Delta K8800 with UBL and Bilinear Levelling. Oh - I just did!

sgparry, 2022-06-30:
The K8800-Blinear is there now, please can we have the UBL variant as well?

taker218, 2022-07-21:
I just got the source code for the X5SA with the STM32F446ZET board but wasn't able to build marlin myself.
Maybe it's helping someone else. It's only from the 2.0.9.2 version of Marlin, not the latest one.
[X5SA-Pro.zip](https://github.com/MarlinFirmware/Configurations/files/9162726/X5SA-Pro.zip)

walterp1982, 2022-07-26:
I got ahold of the Mingda Magician X source code, and verified it compiles. I'd like to get it working on the current Marlin revision and switch to LVGL or TFT Color.

[mingda-1.6.6.5 orig.zip](https://github.com/MarlinFirmware/Configurations/files/9193476/mingda-1.6.6.5.orig.zip)

GRIZZLYGUNNER, 2022-07-28:
Creality Ender 3 pro
btt skr mini e3 v3 board   with btt tft35 e3 v3 screen  ,btt auto power shutdown relay v1.2 and original BL touch v3.1
Please help urgent I've been struggling for more than 3 weeks trying to compile myself but with no luck. I really need to get this 3d printer done as its is someone else's 3d printer. Thanks in advance

walterp1982, 2022-08-01:
I've begun updating the Mingda Firmware to 2.1. I've done a great deal of work, but I'm hitting a wall in resolving some HAL errors during compiling after integrating 2.1 files. I've managed to sort through quite a load, but the recent one is beyond my capability. The challenge is that the board isn't natively supported by Marlin yet, and I'm using Mingda's custom board files. It also uses Extensible UI.

I've uploaded the original mingda source code to github and posted it as a release (1.6.6.5), as well as where I currently stand getting errors, which I've labeled branch 10.0.1. 

The Link: https://github.com/walterp1982/Mingda-Marlin
Attaching the error during compilation
[CompileLog.txt](https://github.com/MarlinFirmware/Configurations/files/9237533/CompileLog.txt)
.

AdamTuraj, 2022-08-05:
Is there an ETA on the Ender S1 Pro configs coming out?

thisiskeithb, 2022-08-05:
> Is there an ETA on the Ender S1 Pro configs coming out?

Someone will need to create, test, and submit them in a pull request.

AdamTuraj, 2022-08-05:
If this someone is me, how would I find the configurations?

thisiskeithb, 2022-08-06:
> If this someone is me, how would I find the configurations?

Ask Creality for a copy of the firmware or find a well-tested community fork that can be ported to a current version of Marlin. Related Ender/Crealtiy Facebook groups may also be able to help out.

EvilGremlin, 2022-08-06:
Or just make one from scratch or closest one, i bet it's faster than asking.

AdamTuraj, 2022-08-06:
> Or just make one from scratch or closest one, i bet it's faster than asking.

I'm not familiar with 3D printer configurations. So that won't work. Unless the S1 configurations will work with the max temp increased.

MasterofNull, 2022-08-11:
If anyone is in need of configuration files for their printer and is not finding one in a timely manner. If you can afford to buy and send me one or do a pool/group purchase. I would be willing to create and post configuration files for you asap. Else I have no way of really knowing if it will work. Wish you all the best.

SteffenKockel, 2022-08-19:
Hi,
I got an DreamMaker Overlord Pro which seems to be supported by Marlin. However, when I compile and then upload the file, the printer becomes unresponsive. The SD-card LED does not light up and the screen stays black. I am able to recover with what seems to be the latest manufacturers firmware. Mainboard version is 2.1.
Can someone help me out here? Has someone managed to run Marlin 2 on this device. What would be needed to add it to the list of supported hardware.
Thanks in advance.

Jhendrix0875, 2022-08-26:
I am looking for help with the new Marlin firmware configs for BTT SKR v1.3 board with a CR TOUCH bed leveling sensor. Any help would be greatly appreciated.

ellensp, 2022-08-26:
@Jhendrix0875 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

Jhendrix0875, 2022-08-26:
> @Jhendrix0875
>
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

I am just asking here as I find the config files for all of the other BTT SKR boards, but the V1.3 and I was just hoping that someone would upload the files in the example folder for future reference....

ProMate04, 2022-09-24:
I would need some configs for a Creality Ender 3 Neo, with CRTouch installed, since I could not find it inbetween the files. I read that if there is no config files for a printer yet, one would need to create, test and commit them here, but I have no idea how I could create them. If someone could send me a guide, it would be really appreciated. Thank you in advance.

ProMate04, 2022-10-03:
> If anyone is in need of configuration files for their printer and is not finding one in a timely manner. If you can afford to buy and send me one or do a pool/group purchase. I would be willing to [...]

I have an Ender 3 NEO, and I would like to be able to create the configs for it so I can customize the firmware. Is there any way you could help me through this process?

MasterofNull, 2022-10-03:
> I have an Ender 3 NEO, and I would like to be able to create the configs for it so I can customize the firmware. Is there any way you could help me through this process?

I can help you. But my time is not free.

Looks like there are many different hardware configurations for this kind of printer. Are you able to figure out which board you have?

Have you looked through the configurations for your printer already?
https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3

Also, there seems to be some uploading caveat potential for your printer. (Please read notes)
https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20V2

You can always try reading through config files and start playing with them yourself.

If you would like to move forward, let's take the convo off this thread.

SangeethsivanSivakumar, 2022-11-02:
Hi, how can I get a config file for Ender 3 V2 Neo?

ellensp, 2022-11-02:
@SangeethsivanSivakumar  I have good and bad new for you.
The good: the motherboard is already supported is a Creality v4.2.2 with tmc2208_standalone stepper drivers
Also I have dumped settings including probe offset and steps/mm etc 
The bad: The new user interface they are so proud of, is not compatible with stock marlin. So the lcd will not work properly yet.

I have attached my notes for this printer thus far.
[ender 3 v2 neo notes.txt](https://github.com/MarlinFirmware/Configurations/files/9923759/ender.3.v2.neo.notes.txt)

ordinarydev, 2022-11-03:
Is there any chance of a config for the Kingroon KP3S Pro printer?

UKIkarus, 2022-11-03:
Hi there, do we happen to have any configs for the mingda magician X to compile the newest marlin firmware?

SangeethsivanSivakumar, 2022-11-03:
> @SangeethsivanSivakumar I have good and bad new for you. The good: the motherboard is already supported is a Creality v4.2.2 with tmc2208_standalone stepper drivers Also I have dumped settings including probe offset and steps/mm etc The bad: The new user interface they are so proud of, is not compatible with stock marlin. So the lcd will not work properly yet.
> 
> I have attached my notes for this printer thus far. [ender 3 v2 neo notes.txt](https://github.com/MarlinFirmware/Configurations/files/9925283/ender.3.v2.neo.notes.txt)
> 
> More bad news They have put in a clone cpu an it a GD32F These are not fully compatible with STM32F very very bad..

Thanks for the detailed reply. I think i'm gonna stick to the stck firmware for now. Octopint works fine for now. So not missing much. Just wanted the mid stop filament function.

NoviceArtisan, 2022-12-05:
Heyo, I'd like to contibute the fruits of my labours this past day, and have the build files adopted into their own pretty spot for everyone to share so that they don't have to loose braincells over trying :)

Fixed the stock Ender 5 Plus LCD to work with SKR 3 and SKR Mini E3 V3 boards. Main issue is a serial port conflict in the source files, with LCD serial and the plain serial both been set to output on the same port. Other issue was mesh size, needed adjusting to 5x5 for the levelling grid to be supported by the DGUS RELOADED LCD firmware. I'm not quite sure how to share the files I've adjusted so you can give them a quick once over before creating a new profile for each of those two boards to work with the stock LCD. If just edits needed, I suggest adding/editing this bit to fix the serial port conflict, tbh it's a bit hard to remember all the changes I did, so probs best to share the configs?

 #if ENABLED(DGUS_LCD_UI_RELOADED)
#define LCD_SERIAL_PORT 1 // LCD will not work without this been set to 1
#define SERIAL_PORT -1 // LCD will conflict with the default SERIAL_PORT 1 given further up in file otherwise
#endif

ellensp, 2022-12-07:
@NoviceArtisan 

Your github repository is not done in the best way.

I looks like You downloaded marlin source (probably as a zip), made your changes and created a new completely separate repository.
There is no easy way to see what change you made to this.

A better way is to fork marlin  (https://github.com/MarlinFirmware/Marlin/fork)

Clone this new fork to you local machine
eg git clone https://github.com/NoviceArtisan/Marlin 

Then create your own branch in your fork (make sure to branch from the correct starting branch, normally bugfix-2.1.x)
eg git checkout bugfix-2.1.x   
git branch SKR3_Ender5Plus

checkout the new SKR3_Ender5Plus branch
git checkout SKR3_Ender5Plus 

Then edit the files in this branch and commit and push/upload them
This creates a repository where people can just click the commits to see what changes you made.
And makes it obvious it is related to the main marlin repository.

NoviceArtisan, 2022-12-07:
> A better way is to fork marlin (https://github.com/MarlinFirmware/Marlin/fork)

Okay, that's very clear and I think I shouldn't have any issue with doing that, barring one teensy issue. When I was attempting to use the Bugfix as is at the time I started, there were a number of bugs that had been introduced, so @thisiskeithb helpfully gave me a direct link to a better working version ( https://github.com/MarlinFirmware/Marlin/commit/5c3d87a951cee94ce3d5ac500079da705a274be4 ), which i was able to use without any weird and unfixable errors been thrown up and work my way through the minor configuration issues present in the codebase without concern. 

I believe that if I clone the current bugfix repo I'll once again face the same issues that bogged me down with my first post (https://github.com/MarlinFirmware/Marlin/issues/25057#)

What would be the correct command to use the repo I was provided with, so I know i can get it working, or indeed if that's okay to use in the same way as the fork process you've described lol :) 

Thank you very much and sorry for the very n00b errors throughout my learning curve!

ellensp, 2022-12-07:
@NoviceArtisan 
From looking at what is changed on your github the only changes are in the Configuration files.
As such I have created a PR to add this to main marlin Configuration examples. https://github.com/MarlinFirmware/Configurations/pull/853

NoviceArtisan, 2022-12-07:
Oh wow @ellensp thank you! ^_^ I do have updated files for the SKR E3 V3 board for the same printer too (it's more of a drop in replacement board that fits perfectly) I don't remember making any adjustments to files other than the configuration ones for that either, pretty sure It's one line of code different to the changes I made to the SKR 3 board (i just changed the motherboard to match and compiled, wired up the lcd to appropriate pins and tested fine) 

When these are added, does it keep the note that these two specific configs require the LCD firmware to be flashed with the desuuuu lcd firmware for marlin, and the bit about the cable pin order? I can edit/comment that into the config files near the start so it's clear if that's neccessary, will only take a minute for me? 

Thank you

ellensp, 2022-12-07:
@NoviceArtisan see https://github.com/ellensp/Configurations/tree/SKR3_Ender5Plus/config/examples/Creality/Ender-5%20Plus/BigTreeTech%20SKR%203

UKIkarus, 2022-12-07:
> Yay, that's so cool! Thank you so much. I feel very good knowing that It'll help so many people to save money and upgrade their printer at lower cost ^_^

Did they mingda magician X ever get ported to 2.1 successfully?

Lost track a bit, apologies.

NoviceArtisan, 2022-12-07:
Whoopsies,  big typo on my part tx2-pa9 rx2-pa10 are mixed up should be tx2-pa10 rx2-pa9, won't work with them reversed (just thankful I didn't fudge the +5v line haha) Could you edit to correct please, thanks :)

ellensp, 2022-12-07:
@NoviceArtisan Done.

Kavci034, 2022-12-18:
Any possibilities of a config for CR10S Pro V2
The DWIN Makes it hard to configure
any way of configuring it without changing the screen since There already is the TH3D Firmware
Creality's official Marlin 1.1.6 firmware files if they would help: https://www.creality.com/pages/download-cr-10s-pro-v2?spm=..page_1934481.products_display_1.1&spm_prev=..page_1934481.header_1.1

DonBower, 2022-12-18:
Looking for Configuration.h and Configuration_adv.h examples for BTT Octopus v1.1

slemson, 2022-12-24:
I don't see 4.2.7 (CrealityV427) example in the Ender-3 V2 folder like I do in, say, the Ender-5 folder. Sorry if this is a noob question but is it as simple as changing the board setting in the config?

EvilGremlin, 2022-12-24:
Board and drivers, and if it's GD chip - goo luck there, you'll need it
Creality is a mess

QuePID, 2022-12-28:
I don't see configuration files for the Bigtreetech SKR-3 board and SKR-3 EZ board. Attached are configuration files I have used for Marlin 2.1.1.

There are significant differences in the SKR 1.4 and SKR 2.0 boards compared to the SKR-3 boards.

[Configuration_SKR3_211.zip](https://github.com/MarlinFirmware/Configurations/files/10312367/Configuration_SKR3_211.zip)

thisiskeithb, 2022-12-28:
> I don't see configuration files for the Bigtreetech SKR-3 board and SKR-3 EZ board

Because these are universal boards that can run any driver type and go into any printer.

Cabbo, 2022-12-29:
Configuration.h and Configuration_adv.h files of Ender3 v1 are too old for Marlin 2.1.2 
Error message: `"Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."`

thisiskeithb, 2022-12-29:
> Configuration.h and Configuration_adv.h files of Ender3 v1 are too old for Marlin 2.1.2

You’ve downloaded the wrong version since the 2.1.2 config for the Ender-3 V1 is fine:

https://github.com/MarlinFirmware/Configurations/blob/7e9b6b7252c72ac5aaebfbd38c246cde4e8b613f/config/examples/Creality/Ender-3/CrealityV1/Configuration.h#L40

https://github.com/MarlinFirmware/Configurations/releases/tag/2.1.2

QuePID, 2022-12-29:
Attached are the configuration.h and configuration_adv.h that I created for the Bigtreetech SKR-3 motherboard with my particular hardware based on the Marlin 2.1.2 generic configuration files.

Hardware includes bltouch, dual z-axis (second z-axis using E1 stepper driver.
[Marlin-2.1.2_config_files.zip](https://github.com/MarlinFirmware/Configurations/files/10320541/Marlin-2.1.2_config_files.zip)

conway220, 2022-12-31:
> @SangeethsivanSivakumar I have good and bad new for you. The good: the motherboard is already supported is a Creality v4.2.2 with tmc2208_standalone stepper drivers Also I have dumped settings [...]

Hey! I saw that you pushed a configuration for the neo v2 to the latest release. Thank you so much for that. Do you have any update on the screen? I was able to use the private folder for the regular v2 but that results in no text showing up under any images. I then used everything in that private folder + a 1.jpg from (I think) stock firmware. That results in me getting text for most things, but some pieces of text are missing.

FL-Phil, 2022-12-31:
Anybody know what I need to change in the configuration to run a RTD PT1000?

Cviatoi, 2023-01-04:
Please add CKR 3 EZ config files for Ender-3 or tell me which board will fit.
Thank you in advance.

DonBower, 2023-01-07:
+1 for Example Creality/Ender-3 V2/MarlinUI with 4.2.7 board.
Bonus points if it's the same config I use for the daily build.

[139 more replies]