MarlinFirmware/Configurations #652: [BUG] (SD Issue)

Enzo-Enz, 2022-01-09:
Marlin doesn't show the file on the sd card 
I use BTT SKR PRO V1.2 and generic delta configuration

thisiskeithb, 2022-01-09:
Without providing the config exactly as modified, I’m going to guess that you forgot to enable `SDSUPPORT` and `SDCARD_CONNECTION` since those are disabled by default.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-01-09 by thisiskeithb

Enzo-Enz, 2022-01-09:
I had enabled SDSUPPORT but when I click on "print from media" It doesn't show the file

Enzo-Enz, 2022-01-09:
[image]
[image]
[image]