MarlinFirmware/Configurations #656: [BUG] (MANUAL_FEEDRATE values must be positive)

MathiasMC, 2022-01-11:
### Bug Description

Cannot compile any version above 2.0.8.2 (Already tested bugfix-2.0.x` branch still same issue) for ender 2 will get an error:

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/7845248/Configuration.zip)

### Steps to Reproduce

1. Download Marlin bugfix-2.0.x` branch
2. Take ender 2 example configs
3. Try to compile

**Expected behavior:**

It should compile

**Actual behavior:**

`In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\Servo.cpp.o] Error 1
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\eeprom.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\eeprom.cpp.o] Error 1
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\watchdog.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\shared\Delay.cpp:25:
Marlin\src\HAL\shared\../../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\AVR\watchdog.cpp.o] Error 1
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\shared\Delay.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../core/../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\shared\../../core/serial.h:24,
                 from Marlin\src\HAL\shared\HAL_MinSerial.h:24,
                 from Marlin\src\HAL\shared\HAL_MinSerial.cpp:22:
Marlin\src\HAL\shared\../../core/../inc/SanityCheck.h:3391:3: error: static assertion failed: MANUAL_FEEDRATE values must be positive.
   static_assert(_PLUS_TEST(8), "MANUAL_FEEDRATE values must be positive.");
   ^~~~~~~~~~~~~
*** [.pio\build\melzi_optiboot_optimized\src\src\HAL\shared\HAL_MinSerial.cpp.o] Error 1`

thisiskeithb, 2022-01-11:
> MANUAL_FEEDRATE values must be positive.

This is not a bug.

ellensp, 2022-01-11:
This does seem to be a bug in the Ender-2 Configuration_adv.h 
ie https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-2/Configuration_adv.h#L1261

```CPP 
#define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 0 } 
```

a 0 value for E feed rate does not make sense. And triggers the assertion

thisiskeithb, 2022-01-11:
> a 0 value for E feed rate does not make sense. And triggers the assertion

2*60 should work.

closed 2022-01-11 by thisiskeithb