MarlinFirmware/Configurations #658: [FR] Tronxy X5ST-500 with CXY-V8-190729

DarkShanox, 2022-01-11:
### Description
My endstops don't seem to be working for my Tronxy X5ST-500 with a CXY-V8-190729 motherboard. I connected with Pronterface while manually pressing the endstops (X, Y and Z) but M119 kept returning 'open'. 
I tried switching between chitu_f103 and chitu_v5_gpio_init but without any success.
Same thing for the board, I tried switching between BOARD_CHITU3D_V5 and BOARD_CHITU3D_V6, without any luck.
I am not quite sure if this issue occurs because I set something incorrectly or if my Tronxy cannot handle Marlin (although I doubt the latter :p ).
Could someone please share the correct Marlin configuration for a X5ST-500 with CXY-V8-190729 motherboard (or have a look at the one I set)?

### Configuration Details

* Manufacturer: Tronxy
* Model: X5ST-500
* Motherboard: cxy-v8-190729

#### Additional Information
I am new to Marlin, since I have my Tronxy X5ST-500 for a while now and never dared to tamper with the firmware. Recently my old motherboard burned down so I bought a cxy-v8-190729 from Tronxy, since they did not have the original one that came with the X5ST-500 anymore. I decided to give Marlin a go and I downloaded the Marlin-tronxy-full-mks-tft files. 

* Provide links to the requested printer/configuration
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/7849010/Configuration.zip)
.

ellensp, 2022-01-12:
According to this https://github.com/MarlinFirmware/Marlin/issues/19320
It is BOARD_CHITU3D_V5

DarkShanox, 2022-01-12:
> According to this [MarlinFirmware/Marlin#19320](https://github.com/MarlinFirmware/Marlin/issues/19320) It is BOARD_CHITU3D_V5

Yeah, I tried setting the board to that one and for the Z-endstop I now have the same phenomenon as was stated here: 
https://github.com/MarlinFirmware/Configurations/issues/321

But for the X and Y stops, M119 keep returning 'open', even though I press them in manually.

closed 2022-01-12 by DarkShanox