MarlinFirmware/Configurations #659: [FR] Tronxy X5ST-500 with CXY-V8-190729 endstops don't seem to work

DarkShanox, 2022-01-12:
### Description

My endstops don't seem to be working for my Tronxy X5ST-500 with a CXY-V8-190729 motherboard. I connected with Pronterface while manually pressing the endstops (X and Y) but M119 kept returning 'open'.
I managed to at least get the 'triggered' response by using the fix mentioned here for the Z-endstop:
https://github.com/MarlinFirmware/Configurations/issues/321

Unfortunately though, the printer does not stop when the Z-endstop is triggered.

Could someone please share the correct Marlin configuration for a X5ST-500 with CXY-V8-190729 motherboard (or have a look at the one I set)?

Configuration Details
Manufacturer: Tronxy
Model: X5ST-500
Motherboard: cxy-v8-190729

### Configuration Details

* Manufacturer: Tronxy
* Model: X5ST-500
* Motherboard: CXY-V8-190729

#### Additional Information

* Provide links to the requested printer/configuration
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/7856833/Configuration.zip)
.

thinkyhead, 2022-01-26:
This Issue Queue is for bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- Marlin Discord server. Join link: https://discord.gg/n5NJ59y
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)

closed 2022-01-26 by thinkyhead

DarkShanox, 2022-01-27:
Sorry, my bad. I thought for Marlin being unable to register an incoming signal was also a bug....