MarlinFirmware/Configurations #665: [FR] Would be helpful to have a configuration for powerstep01 driver (Mega2560 or Due)

mikhail11111, 2022-01-21:
### Description

### Configuration Details

* Manufacturer:
* Model:
* Motherboard:

#### Additional Information

* Provide links to the requested printer/configuration.

ellensp, 2022-01-23:
POWERSTEP01 is already supported
/**
 * Stepper Drivers
 *
 * These settings allow Marlin to tune stepper driver timing and enable advanced options for
 * stepper drivers that support them. You may also override timing options in Configuration_adv.h.
 *
 * A4988 is assumed for unspecified drivers.
 *
 * Use TMC2208/TMC2208_STANDALONE for TMC2225 drivers and TMC2209/TMC2209_STANDALONE for TMC2226 drivers.
 *
 * Options: A4988, A5984, DRV8825, LV8729, L6470, L6474, POWERSTEP01,     <======== see

mikhail11111, 2022-01-23:
The problem is there is no configuration file with POWERSTEP01 in the examples. When I simply insert it instead of A4988 in the default configuration, it doesn't compile with multiple errors.

ellensp, 2022-01-23:
You need to define the pins it uses and a macro
eg for RAMPS

```CPP
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
  #define L6470_CHAIN_SCK_PIN 53
  #define L6470_CHAIN_MISO_PIN 49
  #define L6470_CHAIN_MOSI_PIN 40
  #define L6470_CHAIN_SS_PIN 42
  /**
   * Macro to reset/enable L6474 stepper drivers
   *
   * IMPORTANT - To disable (bypass) L6474s, install the corresponding
   *             resistors (R11 - R17) and change the "V" to "0" for the
   *             corresponding pins here:
   */
  #define ENABLE_RESET_L64XX_CHIPS(V)   do{ OUT_WRITE(X_ENABLE_PIN, V); \
                                          OUT_WRITE(Y_ENABLE_PIN, V); \
                                          OUT_WRITE(Z_ENABLE_PIN, V); \
                                          OUT_WRITE(E0_ENABLE_PIN,V); \
                                          OUT_WRITE(E1_ENABLE_PIN,V); \
                                        }while(0)
#endif
```

ellensp, 2022-01-23:
I don't know what that comment in in referees to, macro is adapted for Marlin/src/pins/stm32f4/pins_STEVAL_3DP001V1.h 
Comment may be motherboard related.

mikhail11111, 2022-01-25:
Now it compiles but still there is no SPI communication with the driver. I suppose the problem is that it needs some additional pins set (probably the Reset and Step clock), but I couldn't find where to do that.

ellensp, 2022-01-26:
The reset is using that macro, and uses the enable pins.

mikhail11111, 2022-01-30:
Right, but still something is missing, not sure what. 
I've been using Marlin only in step-dir mode so far; is there something I might be missing that is pertinent to SPI mode? I connected the SCK, MOSI, MISO, CS.

thinkyhead, 2022-04-10:
Duplicate of #649

closed 2022-04-10 by thinkyhead

Bob-the-Kuhn, 2022-06-19:
The other item that neeeds to be defined is **_CHAIN_POS** for each powerStep.

FYI - I found the TMC5160 to be a superior driver.  I wouldn't mind seeing the L64xx series drivers disapear from Marlin.