MarlinFirmware/Configurations #667: [BUG] Ender 3 4.2.7 config doesnt work in default

T0biasCZe, 2022-01-23:
### Bug Description
The example code for Ender 3 with 4.2.7 motherboard doesnt compile without modifying the example configuration

### Steps to Reproduce
Download Marlin 2.0.9.3
Download these example files
copy over the Ender 3 4.2.7 config files
Open the marlin folder in 

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

The firmware compiles correctly with the unmodified example config

**Actual behavior:**

The firmware throws
`In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2002:6: error: #error "FAN_SOFT_PWM is required. Enable it to continue."
 2002 |     #error "FAN_SOFT_PWM is required. Enable it to continue."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:30:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2002:6: error: #error "FAN_SOFT_PWM is required. Enable it to continue."
 2002 |     #error "FAN_SOFT_PWM is required. Enable it to continue."
      |      ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2002:6: error: #error "FAN_SOFT_PWM is required. Enable it to continue."
 2002 |     #error "FAN_SOFT_PWM is required. Enable it to continue."
      |      ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32\MarlinSerial.cpp.o] Error 1`

thisiskeithb, 2022-01-23:
Enable / uncomment `FAN_SOFT_PWM` for now.

T0biasCZe, 2022-01-23:
Yeah i uncommented it and it compiled... Tho i put the bin on the sd card, it loaded the firmware but it cant auto home... it returns PRINTER HALTED... _should i create separate issue for this?_

ellensp, 2022-01-23:
Check you really have a 4.2.7, installing 4.2.7 firmware on a 4.2.2 machine results in it looking like everything is fin, but no steppers move.

T0biasCZe, 2022-01-23:
Oops yeah its 4.2.2... i incorrectly looked

closed 2022-01-26 by thinkyhead