MarlinFirmware/Configurations #671: [FR]  SKR Mini E3 v3 in combination with CR-10 (stock)

ceesvanegmond, 2022-02-04:
### Description

Hi all. I'm looking for a firmware for the new SKR Mini E3 v3 in combination with CR-10 (stock).
I know I can compile it myself but i'm kind of scared to do it. The machine now has the original Motherboard.
THought upgrading the motherboard was good upgrade but I'm diving a bit in to rabbit hole now.

Can somebody help me out with the config? I know there's config for the V2 (https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0)

Do I only have to change the Motherboard type in the config? Or should there be extra config for this v3?

### Configuration Details

* Manufacturer: Creality
* Model: CR-10
* Motherboard: SKR Mini E3 v3

ellensp, 2022-02-08:
1) Never ever use import branch. it is for developers for updating configs
Use the version of config that is for the version of the code you are using.
Ie if ysing bugfix, use bugfix, if release, use the release
It even says what version to use in your config/README.md file.

2) update MOTHERBOARD is the first thing you do
3) You then need to update the serial ports to match what is plugged in where for the new motherboard
4) update stepper driver types.

this is 99% of it.

Now you compile and fix any errors 
Then you start testing.

ellensp, 2022-02-08:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/649

closed 2022-02-08 by ellensp