MarlinFirmware/Configurations #673: [FR] (configuration request title)

curiousmongrel, 2022-02-07:
### Description

Configuration for Ender 3V2 4.2.7 bpard

### Configuration Details

* Manufacturer:  Creality
* Model: Ender 3V2
* Motherboard: 4.2.7

#### Additional Information

* Provide links to the requested printer/configuration.

*https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2

ellensp, 2022-02-07:
1) Never ever use import branch. it is for developers for updating configs
Use the version of config that is for the version of the code you are using.
Ie if ysing bugfix, use bugfix, if release, use the release   
It even says what version to use in your config/README.md file. 

2) To change from 4.2.2. to 4.2.7 you change one line. That is it.
  #define MOTHERBOARD BOARD_CREALITY_V4 becomes   #define MOTHERBOARD BOARD_CREALITY_V427

ellensp, 2022-02-07:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/649

closed 2022-02-07 by ellensp