MarlinFirmware/Configurations #688: E3 4.2.7 touchscreen. bltouch, runout and 400 tall help

erickson592, 2022-02-28:
I was wondering if anyone is able to change the zaxis to 400mm for the ender 3 firmware with touchscreen, bltouch, and runout sensor.
 I used the one firmware_E3_Tall_Runout_Touch_BLT_DW7.4.5.bin and was great it was just a little short at 350.

thisiskeithb, 2022-02-28:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-02-28 by thisiskeithb

InsanityAutomation, 2022-02-28:
That's one of the precompiled binaries that I maintain on the fork for the configurations from first layer. You can open an issue on my GitHub fork over there or the discord link in order to get small tweaks for it like that.

https://github.com/InsanityAutomation/Marlin/tree/F1rstLayer_Touchscreens