MarlinFirmware/Configurations #704: [FR] 'Break' Import configs deliberately

ellensp, 2022-03-31:
### Description

Despite our best attempts to warn users not to use IMPORT branch on on release or bugfix code they still continue to do so en masse

I suggest that we add something like the following to IMPORT branch Configuration_h and Configuration_adv,h 

#error "This is a IMPORT branch config file. This is not meant to be used  to build your firmware. Please download RELEASE or BUGFIX branch config files"

And auto strip this from RELEASE and BUGFIX when they are generated

thisiskeithb, 2022-04-01:
Making bugfix the default branch here would probably help a lot.

It’d be a minor inconvenience for devs to target the correct branch for PRs, but we already do something like this in the main Marlin repo.

thinkyhead, 2022-04-10:
I suppose I can add an `#error` line to the top of `Configuration.h` in the `import-2.0.x` branch, which is stripped out by the `mfconfig` script that consolidates them for the generated branches.

closed 2022-04-10 by ellensp