MarlinFirmware/Configurations #711: [FR] (draw_x_mode in idex, TFT LVGL)

Meu88, 2022-04-11:
### Description

Sorry for my bad english, its my firts question and it possible that make so bad, sorry in advance. I think that file draw_x_mode is missing in \Marlin\src\lcd\extui\mks_ui, i have a artist-d Pro and all is fine in 2.9.0.3 but dont show this option in TFT_lvgl mode.

### Configuration Details

* Manufacturer: JgMaker
* Model: Artist-D Pro
* Motherboard: Robin_mks_pro v1.0

#### Additional Information

i have compared the files with original version from jgmaker source in v.2.0.5

* Provide links to the requested printer/configuration.

thisiskeithb, 2022-04-20:
Do you have photos of what this draw_x_mode screen looks like? Are there any other IDEX-specific screens or icons added for this printer?

The [ArtistD-Pro repo](https://github.com/JGMaker3dofficial/ArtistD-Pro) is using Marlin `2.0.5.3` which is pretty old at this point, so it'd need to be brought up to date & tested with the latest MKS UI code since it's been refactored & cleaned up since then.

Meu88, 2022-04-20:
Yes, when the vscode is compiling with tft-lvgl mode, vscode go to web page and download "assets", in this assets appear like bmp_x_mode.bin, but in the route (\Marlin\src\lcd\extui\mks_ui) not appear the instrucction. 

There are some icons, in original look like the attachement png
[image]

in the asset version from download its diferent but i cant see in Makerbase Image editor because  I havent the config file.

For now, i dont now if any other specific screnns is necessary, but i lose that the picture from extrude 2 have to be move to right because it is on top of the value of extruder 1, in any case, I can review and say what I miss with the options that I have activated, for now without bltouch, take photos and more if is necessary, now i have installed tft-color but i can reinstall and see if anything else is missing

in case it's interesting, when you have perfectly calibrated z axis in both extruder and you make babysteps in extruder 1, the extruder 2 dont be afeccted for this babysteps, this forces you to have to recalibrate the z-axis every time you play the babysteps in the same proportion.

Again, sorry for my bad english.

thisiskeithb, 2022-08-04:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23318

closed 2022-08-04 by thisiskeithb