MarlinFirmware/Configurations #713: Sunlu S8 configuration does not enable hotend heat sink fan.[BUG] (short description)

stefanhra, 2022-04-13:
### Bug Description

The Sunlu S8 printer uses a PWM pin to control the hotend heat sink fan which turns on and off based on extruder temperature. The configuration files provided here do not enable this PWM control for the heat sink fan so it never turns on. The configuration.h file provided by the manufacturer for Marlin 1.x uses MOTHERBOARD BOARD_MKS_BASE while the configuration.h here uses MOTHERBOARD MOTHERBOARD BOARD_RAMPS_14_EFB.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Upload Marlin firmware to Sunlu S8 printer using configuration files from this project.
2. Set temperature of hotend to printing temperature (eg. 220 C)
3. Observe hotend heat sink fan. 

**Expected behavior:**

Hot end heat sink fan should start running when hot end heats up. 

**Actual behavior:**

Hot end heat sink fan doesn't run. 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.
[Sunlu S8.zip](https://github.com/MarlinFirmware/Configurations/files/8484832/Sunlu.S8.zip)

thisiskeithb, 2022-04-18:
According to @InsanityAutomation https://github.com/MarlinFirmware/Configurations/pull/512#issuecomment-1101377824, Sunlu must have shipped some of these out with RAMPS boards and MKS Base boards, so I'll create a new config.

InsanityAutomation, 2022-04-18:
The board change alone does not define an auto fan pin. The MKS base simply sets the board name then includes the ramps file, so functionally irrelevant. The auto fan pin would be defined in config adv. I did a quick bit of google fu, and it seems the config published has this line added : 
#define E0_AUTO_FAN_PIN 7

Which on machines newer than mine which was wired right to power would add the functionality you described.

thisiskeithb, 2022-04-18:
> The board change alone does not define an auto fan pin. The MKS base simply sets the board name then includes the ramps file, so functionally irrelevant.

I’m well aware of this. It’s not my first config 😛 

I’m not sure why you’re so adverse to adding a new / proper config for machines that shipped differently than yours. It can easily live with the current config just like we do for every other printer we have here.

InsanityAutomation, 2022-04-18:
No need for the extra config to define that line in the existing one IMO. If it has the auto fan wired, that line will use it. If it doesnt, itll happily blink that pin with nothing attached. If the board name is desired to be MKS Base instead of Ramps, I also dont have a preference on the existing one being changed or left, as its nothing but a name.

So IMO the simplest resolution is to add the auto fan line to the existing config, and everything works for both happily and users dont have to guess what they have.

InsanityAutomation, 2022-04-18:
> 
> I’m well aware of this. It’s not my first config 😛
> 

That was more to explain it to the OP than you btw, i know youve done plenty lol

thisiskeithb, 2022-04-18:
@stefanhra: Can you take a photo of the motherboard in your printer?

stefanhra, 2022-04-19:
@thisiskeithb Yes, I will post a picture my motherboard this evening.

@InsanityAutomation Thanks for the info. I managed to muddle my way through by comparing the config files provided by Sunlu with the ones provided here.

thisiskeithb, 2022-04-19:
@stefanhra Does setting `#define E0_AUTO_FAN_PIN 7` work for your printer with the Sunlu S8 config provided here & `MOTHERBOARD` defined as `BOARD_RAMPS_14_EFB`? I checked the RAMPS pins and I'm not sure if it'd work/may conflict with other features.

stefanhra, 2022-04-19:
[image]

@thisiskeithb here is a picture of the motherboard in my  Sunlu S8. Give me a bit and I'll try defining the motherboard as BOARD_RAMPS_14_EFB and check if the fan works.

stefanhra, 2022-04-19:
@thisiskeithb the fan does work with #define MOTHERBOARD BOARD_RAMPS_14_EFB. My two cents on the subject is that #define MOTHERBOARD BOARD_MKS_BASE is nice because then you don't need to choose if the motherboard is using an ATmega 2560 or ATmega 1280.

thisiskeithb, 2022-04-19:
I would add a new motherboard and call it `SH 2560_A4988 V1`, but not everyone agrees with that. You could then have a proper motherboard name and skip all this confusion about pins/board type.

stefanhra, 2022-04-19:
That's actually a pretty good idea because motherboards don't always stay with their original printers. Actually I intend to upgrade my S8 with a 32 bit board eventually and this motherboard would be used to replace a RAMPs 1.4 on another printer in that case. Maybe you could "just" fork this project and maintain "MarlinFirmware_by_motherboard".

In any case I'm just glad to have 99.999% working config files to start from. Marlin has added a LOT of new features since I last configured a printer from scratch.

thisiskeithb, 2022-04-21:
See https://github.com/MarlinFirmware/Configurations/pull/718

closed 2022-04-21 by thisiskeithb