MarlinFirmware/Configurations #715: [FR] Set bugfix-2.0.x branch as default

thisiskeithb, 2022-04-19:
### Feature Request Description

Despite our best efforts to prevent compiling with the `import-2.0.x` branch, users are still trying to run it due to it being the default branch when visiting this repo.

Can the `bugfix-2.0.x` branch be set as default for this repo?

The only “issue” is that developers would need to target the correct branch when submitting PRs, but we that’s already expected in the main Marlin repo and we have a bot to handle PRs pointed at the incorrect branch with instructions.

thisiskeithb, 2022-04-19:
Also, is https://github.com/MarlinFirmware/Configurations/tree/master used at all? That can probably get purged.

berkeoznalbant, 2022-04-21:
Yeah, I accidentally used import branch when I bought a new skr board. I was just learning how to compile and flash firmware.

thinkyhead, 2022-04-24:
It would make slightly more sense to have a branch called "release" as the default, and have it aliased to the current release.

closed 2022-04-24 by thinkyhead