MarlinFirmware/Configurations #736: [BUG] Arduino IDE/Anycubic Kossel Config error: 'manual_feedrate_mm_s' was not declared in this scope

rudowinger, 2022-05-15:
### Bug Description

Unable to compile/verify the "Anycubic Kossel" configuration.h / configuration_adv.h with the corresponding version
However, the delta/generic example compiles fine

**The error in detail:**
(THE SKETCH FOLDER)\Marlin\src\lcd\marlinui.cpp: In static member function 'static void ManualMove::task()':
(THE SKETCH FOLDER)\Marlin\src\lcd\marlinui.cpp:796:60: error: 'manual_feedrate_mm_s' was not declared in this scope

         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;

(THE SKETCH FOLDER)\Marlin\src\lcd\marlinui.cpp:796:60: note: suggested alternative: 'feedrate_mm_s'

         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;

[Full error log.txt](https://github.com/MarlinFirmware/Configurations/files/8695878/Full.error.log.txt)

**Environment:**
Marlin fresh 2.0.9.3
Configuration release freh 2.0.9.3
Arduino IDE 1.9.19 on Win 10
Board: Arduino Mega or Mega 2560
Processor: ATMega2560 (2560)
Programmer: AVRISP mkII

### Configuration Files
[fresh anycubic.zip](https://github.com/MarlinFirmware/Configurations/files/8695853/fresh.anycubic.zip)
Unmodified, fresh from Configurations-release-2.0.9.3.zip

### Steps to Reproduce

1. Copy over the Anycubic Kossel config example file from the 2.0.9.3 ZIP into fresh Marlin 2.0.9.3
2. Open Arduino IDE by doubleclicking the Marlin.ino
3. Click "Verify"

**Expected behavior:**
"Sketch compiled"

**Actual behavior:**
"Error compiling for board Arduino Mega or Mega 2560"

#### Additional Information
I tried everything, multiple just values-popping into confirmed-working config files etc., cannot find what causes the error

thisiskeithb, 2022-05-16:
You'll need to use Marlin from the tagged release instead: https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.3.zip

rudowinger, 2022-05-17:
> The linked `2.0.x` branch on marlinfw.org is a rolling branch with updates from bugfix, so that's where this error comes from.
> 
> You'll need to use Marlin from the tagged 2.0.9.3 release until the next release is tagged: https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.3.zip

Wow, thank you so much!!! I guess I would never found the issue without you!

**Managed to compile with the following combination of Marlin/Confs 2.0.9.3:**
Marlin: https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.3.zip
Configuration: https://github.com/MarlinFirmware/Configurations/archive/refs/heads/release-2.0.9.3.zip

Suggestion for the Marlin website: Maybe make new/returning users' lives easier by offering Marlin/Configurations bundles (Firmware/Conf combinations that compile fine) of stable versions?

[image]

Thank you! Was still on 1.0, looking forward to use all the new features of Marlin 2.0!

closed 2022-05-17 by rudowinger

thisiskeithb, 2022-05-17:
> Suggestion for the Marlin website: Maybe make new/returning users' lives easier by offering Marlin/Configurations bundles (Firmware/Conf combinations that compile fine) of stable versions?

Normally, there'd be more frequent `2.0.x` releases, but there have been a lot of PRs & issues to get through. It should be fixed once the next release is tagged 🙂

frederic76430, 2022-05-17:
bonsoir
j ai ce message erreur aussi avec la2.0.9.3 mais pas avec la bugfix
merci

Compiling .pio\build\mega2560\src\src\lcd\menu\menu_advanced.cpp.o
Marlin\src\lcd\marlinui.cpp: In static member function 'static void ManualMove::task()':
Marlin\src\lcd\marlinui.cpp:796:60: error: 'manual_feedrate_mm_s' was not declared in this scope
         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;
                                                            ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\marlinui.cpp:796:60: note: suggested alternative: 'feedrate_mm_s'
         const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S;
                                                            ^~~~~~~~~~~~~~~~~~~~
                                                            feedrate_mm_s
Compiling .pio\build\mega2560\src\src\lcd\menu\menu_bed_corners.cpp.o
*** [.pio\build\mega2560\src\src\lcd\marlinui.cpp.o] Error 1
Marlin\src\lcd\menu\menu.cpp: In function 'void line_to_z(const_float_t)':
Marlin\src\lcd\menu\menu.cpp:289:30: error: 'manual_feedrate_mm_s' was not declared in this scope
     line_to_current_position(manual_feedrate_mm_s.z);
                              ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.cpp:289:30: note: suggested alternative: 'feedrate_mm_s'
     line_to_current_position(manual_feedrate_mm_s.z);
                              ^~~~~~~~~~~~~~~~~~~~
                              feedrate_mm_s
*** [.pio\build\mega2560\src\src\lcd\menu\menu.cpp.o] Error 1
Marlin\src\lcd\menu\menu_bed_corners.cpp: In function 'void _lcd_goto_next_corner()':
Marlin\src\lcd\menu\menu_bed_corners.cpp:316:30: error: 'manual_feedrate_mm_s' was not declared in this scope
     line_to_current_position(manual_feedrate_mm_s.x);
                              ^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_bed_corners.cpp:316:30: note: suggested alternative: 'feedrate_mm_s'
     line_to_current_position(manual_feedrate_mm_s.x);
                              ^~~~~~~~~~~~~~~~~~~~
                              feedrate_mm_s
*** [.pio\build\mega2560\src\src\lcd\menu\menu_bed_corners.cpp.o] Error 1

rudowinger, 2022-05-17:
> bonsoir j ai ce probleme avec mon geetech A20m..
> 
> j ai ce message erreur aussi avec la2.0.9.3 mais pas avec la bugfix merci
> 
> Compiling .pio\build\mega2560\src\src\lcd\menu\menu_advanced.cpp.o Marlin\src\lcd\marlinui.cpp: In static member function 'static void ManualMove::task()': Marlin\src\lcd\marlinui.cpp:796:60: error: 'manual_feedrate_mm_s' was not declared in this scope const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S; ^~~~~~~~~~~~~~~~~~~~ Marlin\src\lcd\marlinui.cpp:796:60: note: suggested alternative: 'feedrate_mm_s' const feedRate_t fr_mm_s = (axis < LOGICAL_AXES) ? manual_feedrate_mm_s[axis] : XY_PROBE_FEEDRATE_MM_S; ^~~~~~~~~~~~~~~~~~~~ feedrate_mm_s Compiling .pio\build\mega2560\src\src\lcd\menu\menu_bed_corners.cpp.o *** [.pio\build\mega2560\src\src\lcd\marlinui.cpp.o] Error 1 Marlin\src\lcd\menu\menu.cpp: In function 'void line_to_z(const_float_t)': Marlin\src\lcd\menu\menu.cpp:289:30: error: 'manual_feedrate_mm_s' was not declared in this scope line_to_current_position(manual_feedrate_mm_s.z); ^~~~~~~~~~~~~~~~~~~~ Marlin\src\lcd\menu\menu.cpp:289:30: note: suggested alternative: 'feedrate_mm_s' line_to_current_position(manual_feedrate_mm_s.z); ^~~~~~~~~~~~~~~~~~~~ feedrate_mm_s *** [.pio\build\mega2560\src\src\lcd\menu\menu.cpp.o] Error 1 Marlin\src\lcd\menu\menu_bed_corners.cpp: In function 'void _lcd_goto_next_corner()': Marlin\src\lcd\menu\menu_bed_corners.cpp:316:30: error: 'manual_feedrate_mm_s' was not declared in this scope line_to_current_position(manual_feedrate_mm_s.x); ^~~~~~~~~~~~~~~~~~~~ Marlin\src\lcd\menu\menu_bed_corners.cpp:316:30: note: suggested alternative: 'feedrate_mm_s' line_to_current_position(manual_feedrate_mm_s.x); ^~~~~~~~~~~~~~~~~~~~ feedrate_mm_s *** [.pio\build\mega2560\src\src\lcd\menu\menu_bed_corners.cpp.o] Error 1

Bonjour!

J'ai eu la même erreur, vous pouvez essayer de configurer votre Marlin/Configuration avec la combinaison suivante :

Marlin: https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.3.zip
Configuration: https://github.com/MarlinFirmware/Configurations/archive/refs/heads/release-2.0.9.3.zip

Assurez-vous de repartir à zéro ! Ne copiez pas de nouveaux fichiers sur d'anciens vones, mais créez le dossier MArlin frais et placez-y les nouveaux fichiers de configuration. Il devrait maintenant compiler :)

(J'ai utilisé une nouvelle copie de tout, obtenu le dernier IDE Arduino et mis à jour toutes les bibliothèques
J'espère que cela t'aides!

frederic76430, 2022-05-17:
Arduino ide j'utilise visual avec pio

rudowinger, 2022-05-17:
> Arduino ide j'utilise visual avec pio

Essayez la combinaison de fichiers que j'ai suggérée, peut-être que cela fonctionne aussi pour pio