MarlinFirmware/Configurations #742: [BUG] compilation fails for Two Trees Bluer Plus with mks_robin_nano35_maple

Stefan-van-Hattum, 2022-05-29:
### Bug Description

Downloaded latest Marlin 2.0.9.3 and the configurations file 2.0.9.3
After making the following modifications according to YouTube channel Aurora Tech:

#define Z_MIN_ENDSTOP_INVERTING false// Set to true to invert the logic of the endstop.

// XPT2046 Touch Screen calibration
#define TOUCH_CALIBRATION_X 16903
#define TOUCH_CALIBRATION_Y 10346
#define TOUCH_OFFSET_X        -25
#define TOUCH_OFFSET_Y        -22

I saved the config and try to build using mks_robin_nano35_maple.
The following is shown:

> Executing task: C:\Users\Stefan\.platformio\penv\Scripts\platformio.exe run --environment mks_robin_nano35_maple <

Processing mks_robin_nano35_maple (board: genericSTM32F103VE; platform: ststm32@~12.1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103VE.html
PLATFORM: ST STM32 (12.1.1) > STM32F103VE (64k RAM. 512k Flash)       
HARDWARE: STM32F103VET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple @ 3.10000.201129 (1.0.0)
 - tool-stm32duino @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SoftwareSerialM @ 1.0.0
|-- STM32ADC @ 1.0
|-- EEPROM
|-- USBComposite for STM32F1 @ 0.99
|-- Wire @ 1.0
|-- Servo(STM32F1) @ 1.1.2
Building in release mode
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\Print.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\WString.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\hooks.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\itoa.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\mks_robin_nano35_maple\FrameworkArduino\libmaple\gpio.c.o
[... 217 lines cut ...]

If you've made any other modifications describe them in detail here.

[configurations.zip](https://github.com/MarlinFirmware/Configurations/files/8793183/configurations.zip)

### Steps to Reproduce

1. Make modifcation in Configuration.h
2. Save modified Configuration.h
3. Use PlatformIO and select mks_robin_nano35_maple
4. click build
5. wait for error to apear

**Expected behavior:**

I expect compilation successful and have a custom Marlin firmware for my Two Trees Bluer Plus.

**Actual behavior:**

Build fails with errors

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2022-05-29:
You did not download Marlin 2.0.9.3 https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.3.zip 
You downloaded Marlin 2.0.x ie https://github.com/MarlinFirmware/Marlin/archive/2.0.x.zip

2.0.x is slightly ahead of 2.0.9.3 and breaks if you use 2.0.9.3 config files.

In particular In Configuration_adv.h this has changes a lot recently

 2.0.x needs 
```
// @section lcd

#if HAS_MANUAL_MOVE_MENU
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel
```

Marlin 2.0.9.2 - 2.0.9.3 needs
```
// @section lcd

#if ANY(HAS_LCD_MENU, EXTENSIBLE_UI, HAS_DWIN_E3V2)
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel
```

closed 2022-05-29 by ellensp

Stefan-van-Hattum, 2022-10-11:
Hi ,

Thank you for the response, looking at the website of Marlin it says version 2.0.9.3 but the link is indeed 2.0.x.zip which I didn’t notice.
That is a mistake on the website I guess then.
***@***.***

I will try to download the 2.0.9.3 from github using the link you supplied.
Thank you !.

Regards,

Stefan