MarlinFirmware/Configurations #750: [BUG] FoamCutter example config compile error

DerAndere1, 2022-06-10:
### Bug Description

FoamCutter example config cannot be compiled.

### Configuration Files

[FoamCutter example config link
](https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.1.x/config/examples/FoamCutter/Configuration.h )

### Steps to Reproduce

1. Use FoamCutter example config file, compile corresponding branch of Marlin 
2. compile corresponding branch of Marlin 

**Expected behavior:**

Successfully compile 

**Actual behavior:**

Compile error because it enabled `FOAMCUTTER_XYUV` in Configuration.h which requires 5 axes (`I_DRIVER_TYPE` and J_DRIVER_TYPE etc.)

#### Additional Information

None

thisiskeithb, 2022-06-10:
Yep. It's one of many: https://github.com/MarlinFirmware/Configurations/pull/737#issuecomment-1128308417

DerAndere1, 2022-06-22:
Closing as duplicate of issue 737. Also, since https://github.com/MarlinFirmware/Configurations/pull/751 and https://github.com/MarlinFirmware/Marlin/pull/24325 were merged, there is a working example config for a well documented foam cutter: https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/FoamCutter/rcKeith

closed 2022-06-22 by DerAndere1