MarlinFirmware/Configurations #753: Marlin 2.1 - MKS Robin Nano V3.1 + MKS TS35 + MKS PWC module

fludziu, 2022-06-17:
When I want to compile latest version of Marlin 2.1 with MKS Robin Nano V3.1 motherboard and MKS TS35 display, so do with MKS PWC module I'm not able to do that.
I have following error:
`error: #error "PSU_CONTROL is incompatible with MKS_PWC plus TFT_LVGL_UI."`
Before I was using Marlin 2.0.9.3 and there were no issues with this configuration:
```
#define PSU_CONTROL
#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define MKS_PWC                 // Using the MKS PWC add-on
  #define PS_OFF_CONFIRM          // Confirm dialog when power off
  #define PS_OFF_SOUND            // Beep 1s when power off
  #define PSU_ACTIVE_STATE HIGH 
```

Another issue which I still have in latest Marlin version, so do previously compiled is turning on filament sensor, when I activate in configuration and configuration_adv
`#define FILAMENT_RUNOUT_SENSOR`
`#define ADVANCED_PAUSE_FEATURE`

I receive following errors:
```
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp: In function 'void btn_ok_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:137:7: error: 'pause_menu_response' was not declared in this scope; did you mean 'pause_menu_disp'?
  137 |       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
      |       ^~~~~~~~~~~~~~~~~~~
      |       pause_menu_disp
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:137:29: error: 'PAUSE_RESPONSE_EXTRUDE_MORE' was not declared in this scope
  137 |       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\lcd\extui\mks_ui\../../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:27:
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp: In function 'void btn_cancel_event_cb(lv_obj_t*, lv_event_t)':
Marlin\src\lcd\extui\mks_ui\draw_dialog.cpp:189:35: error: 'pause_menu_response' was not declared in this scope; did you mean 'pause_menu_disp'?
  189 |     TERN_(ADVANCED_PAUSE_FEATURE, pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT);
```
I'm using graphic UI.

Could you help me to solve this issue?

In attachment there are my files from 2.1 and 2.0.9.3 config
[MKS_PWC_issue.zip](https://github.com/MarlinFirmware/Configurations/files/8930966/MKS_PWC_issue.zip)
.

thisiskeithb, 2022-06-17:
See https://github.com/MarlinFirmware/Marlin/issues/23698

closed 2022-06-17 by thisiskeithb

fludziu, 2022-06-17:
Ok, thanks for feedback! So I just back to setup from 2.0.9.3...

And what about filament sensor, how I may run it on?

ysnuysl, 2023-12-27:
Were you able to resolve the error for the filament sensor?