MarlinFirmware/Configurations #755: [BUG] Acceleration values for Anycubic Chiron to high

elFallino, 2022-06-18:
**Problem:**

The _Acceleration Values_ for the Anycubic Chiron are configured as the following:
```
Max Acceleration (units/s2):
M201 X3000.00 Y3000.00 Z100.00 E10000.00
Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
M204 P3000.00 R3000.00 T3000.00
```
This causes heavy mechanical stress and massive layer-shifts at least on the Z-axis as the Chiron uses a heavy 400x400mm bed.

<br>

**Solution:**

Lowering the Acceleration Values.

The Vendors original settings (taken from the latest original Firmware) are working fine.
The Ventor configured them as follows:

```
Maximum Acceleration (mm/s2):
M201 X350 Y350 Z50 E20000
Accelerations: P=printing, R=retract and T=travel
M204 P350.00 R2000.00 T350.00
```

Maybe values in between would work, too but I have not testet them yet.

thisiskeithb, 2022-07-25:
Fixed in https://github.com/MarlinFirmware/Configurations/pull/788

closed 2022-07-25 by thisiskeithb

ntbarone, 2024-11-23:
I'm sorry. I am not a coder, and I'm trying to get my Chiron in working order. Where do I copy this to in my config?
Thank you,
Nino