MarlinFirmware/Configurations #756: M122 S0 doesn't stop or prevent continuous debugging output

CBoismenu, 2022-06-19:
### Bug Description

With TMC_DEBUG enabled in Configuration_adv.h using the M122 S0 command doesn't prevent continuous debugging from being trigered

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Compile firmware
2. M122 S0

**Expected behavior:**

Expecting printer not to enter Continuous Debugging Mode.

**Actual behavior:**

M122 S0 or M122 S1 produces the same output system enters Continuous Debugging Mode

#### Additional Information
Output after M122 S0 command:

Send: M122 S0
Recv: axis:pwm_scale/curr_scale/mech_load|flags|warncount
Recv: 		X	Y	Z	Z2	E
Recv: Address		0	0
Recv: Enabled		false	false	true	true	false
Recv: Set current	550	550	200	200	550
Recv: RMS current	520	520	183	183	520
Recv: MAX current	733	733	258	258	733
Recv: Run current	16/31	16/31	5/31	5/31	16/31
Recv: Hold current	8/31	8/31	2/31	2/31	8/31
Recv: CS actual	8/31	8/31	2/31	2/31	8/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: interp		true	true	true	true	true
Recv: tstep		max	max	max	max	max
Recv: PWM thresh.	98	98	98	98	188
Recv: [mm/s]		100	100	20	20	30
Recv: OT prewarn	false	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false	false
Recv: pwm scale sum	12	11	11	10	10
Recv: pwm scale auto	0	0	0	1	0
Recv: pwm offset auto	45	42	127	106	36
Recv: pwm grad auto	13	12	25	23	14
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2	2
Recv:  -start		1	1	1	1	1
Recv: Stallguard thrs	60	60
Recv: uStep count	408	424	360	824	88
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: sg_result	60	110
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:08:00:00
Recv: 		Y	0xC0:08:00:00
Recv: 		Z	0xC0:02:00:00
Recv: 		Z2	0xC0:02:00:00
Recv: 		E	0xC0:08:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok
Recv: X:12/8/-|TI|	Y:11/8/-|TI|	Z:11/2/-|TI|	Z2:10/2/-|TI|	E:10/8/-|TI|
Recv:  T:150.06 /150.00 B:60.00 /60.00 @:26 B@:0
Recv: Not SD printing
Recv: X:12/8/-|TI|	Y:11/8/-|TI|	Z:11/2/-|TI|	Z2:10/2/-|TI|	E:10/8/-|TI|
Recv: X:0.0000 Y:0.0000 Z:15.0000 E:0.0000 Count X:0 Y:0 Z:6023
Recv: X:12/8/-|GTI|	Y:11/8/-|GTI|	Z:11/2/-|GTI|	Z2:10/2/-|GTI|	E:10/8/-|GTI|
Recv: Not SD printing
Recv: X:12/8/-|TI|	Y:11/8/-|TI|	Z:11/2/-|TI|	Z2:10/2/-|TI|	E:10/8/-|TI|

* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/8936048/Marlin.zip)

CBoismenu, 2022-06-19:
This is with Marlin 2.1 and the latest bugfix version

thisiskeithb, 2022-06-19:
Please file a proper bug report in the main Marlin Firmware repo: https://github.com/MarlinFirmware/Marlin

closed 2022-06-19 by thisiskeithb