MarlinFirmware/Configurations #764: tengo problema con la configuracion de un lcd mks mini 12864 v3[BUG] (short description)

rcampa68, 2022-06-24:


rcampa68, 2022-06-24:
me surge un error al compilar para una placa mks monster 8 una pantalla led sks mini 12864 v3 el cual es el siguiente ya estoy desesperado

#error "Your FYSETC/MKS/BTT Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB."

alguien sabe como remediarlo

ellensp, 2022-06-24:
Do as it says....

Find NEOPIXEL_TYPE in Configuration.h
change the value to NEO_RGB  from the default NEO_GRBW 

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-06-24 by ellensp

rcampa68, 2022-06-25:
Ya pude gracias al consejo pasar ese error pero me salen todavia estas dos advertencias como podria quitarlas?

    #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"

#warning "Maple build environments are deprecated. Please use a non-Maple build environment. Report issues to the Marlin Firmware project."

rcampa68, 2022-06-25:
#warning "Your FYSETC/MKS/BTT Mini Panel works best with NEOPIXEL_LED, LED_CONTROL_MENU, LED_USER_PRESET_STARTUP, and LED_COLOR_PRESETS."