MarlinFirmware/Configurations #767: TINA2 babystep

atsju, 2022-06-26:
Good day I have the tina 2 verison now tested problem is the axes can not be set via the display z-offset so unfortunately not possible baby steps are also not possible do you have a solution?

_Originally posted by @xxoconenrxx in https://github.com/MarlinFirmware/Configurations/issues/685#issuecomment-1166443705_

atsju, 2022-06-26:
Please explain your problem further and share the exact FW and configuration version. 
Babystep is available in menu tune->babystepZ. Do you mean Babystep is not changing heights OR the offset is not saved between power ups ?

xxoconenrxx, 2022-06-26:
Good day I use the firmware marlin 2.1.x with the config from the weedo tina 2 v3 ornder problem is the display has the function z-offset with the rotary knob, the motors do not respond to the change on the rotary knob when I said in the oringale z-offset and turn the knob drove the motor now it is not in the version whether a setting in marlin is missing or disabled . simply said the motor does not respond to commands from the z-offset in the display 

Translated with www.DeepL.com/Translator (free version)

atsju, 2022-06-26:
2.1.x is not a fixed version. Tomorrow 2.1.x is not same as todays 2.1.x. You need to share the tag (2.1.0) or the sha (see picture)
[image]

Does the Z motor move in "move axis" menu ? does Z motor move when doing "auto home" ?
What menu exactly do you use ? Please provide each step to reproduce.

When using translator, please do very short sentences. It translates better. It is clearer for non-english speakers.

xxoconenrxx, 2022-06-26:
es homen works, i go to bed level and want to set my z offset but this does not work because the motor does not move when setting the z offset in the display menu

thisiskeithb, 2022-06-26:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-06-26 by thisiskeithb

xxoconenrxx, 2022-06-26:
i am using the 2.1.0 with the weedo tina 2 v3 i have also tried other marlin versions but all the same error. no motor on z-offset movement when i want to set it in the display

xxoconenrxx, 2022-06-26:
@thisiskeithb  unfortunately the links don't get me anywhere either. i have a marlin verison here that works. but i can't use it like this i've already used the links to find the error

atsju, 2022-06-26:
We can still continue discussion here. Issue is just closed because it's not a marlin bug. 

So you confirm you only problem is : "motor does not move on babybstep" ? Are you sure it does not move at all? 0.001 is 1/1000th mm you will not "see" The motor moving. Please move 1.000 and confirm motor did not move. 

Please share you configuration files here because you did not share version I requested for configuration.

xxoconenrxx, 2022-06-26:
yes i know that, but it is also in the z-offset everything i want to adjust what with z-offset or addendum (babyset) does not work. i have been using marlin for years i am aware of all this. unfortunately i cannot find the trigger here as it does not turn any motors via the display

xxoconenrxx, 2022-06-26:
Here the data

thisiskeithb, 2022-06-26:
> We can still continue discussion here.

No, this is not the place.

Please use one of the resources I linked above for configuration issues.

atsju, 2022-06-26:
@thisiskeithb the issue is closed. I don't see the problem continuing here (no personal offense, just I don't see). Don't you have bigger problems to solve? 
Moreover levelling is working but babbystep is not. This could sound like a FW issue (or not but I could easily open an issue that you would not close. It's a bit subjective) 

@xxoconenrxx please open issue there https://github.com/atsju/wikiTest/issues it's a random personal repo. Make sure to share your configuration files so I can try to reproduce. After I personally test we will see how to proceed.

thisiskeithb, 2022-06-26:
> the issue is closed. I don't see the problem continuing here (no personal offense, just I don't see). Don't you have bigger problems to solve?

You are sending notifications to anyone following MarlinFirmware as well as our Discord. Please continue troubleshooting/tech support in the resources I linked above in my previous comment.

xxoconenrxx, 2022-06-27:
yes i will try another way but what you say is not right the leveling does not work and the bebystepping as well as the stepping of the z-offset