MarlinFirmware/Configurations #771: [BUG] K8800 bed levelling default inconsistent and hazardous.

sgparry, 2022-07-03:
### Bug Description

The newly added configuration for the K8800 has a default bed levelling that is inconsistent, confusing and possibly hazardous.
The original Pull Request included two configurations - one for bilinear leveling, one for UBL. On commit, these were merged together using `#define` and `#if`, but this introduced some problems:

- the `#define` default is currently UBL, whereas the folder is called `K8800-Bilinear`; Bilinear would be the safer default (UBL requires the printer definition in the slicer to be changed; failure to do so can cause bed-to-head crashes.).
- The UUID for the variants is the same - Different UUIDs were initially defined to allow a slicer / host to quickly differentiate between the two variants.
- Anyone wanting to use the non-default variant will have to build it themselves; many users rely on automated build services to update their Marlin firmware.
It did however have the advantage of only adding one profile to Marlin rather than two.

### Configuration Files
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/9035230/Configuration.zip)

### Steps to Reproduce

- Build 2.1.1 firmware using the K8800-Bilinear configuration
- Upload to your K8800
- Slice a test model with Cura using default settings for K8800
- Print the model.
or
- look in Configuration.h at around line 31 and compare `#define K8800_*` settings with the name of the folder
- look in Configuration.h at around line 154 at the UUID `#define`.

**Expected behavior:**

Default configuration should build as bilinear levelling, or as both bilinear and UBL. The bilinear variant should work as a drop in replacement for the stock 1.1.4 firmware.

**Actual behavior:**

The print will either stall with an error or head crash just after the purging part of preflight.

#### Additional Information

I will submit two mutually exclusive PRs to fix this - one for a combined configuration, one for split. The project owners are free to adopt whichever one (or none) they please.

thinkyhead, 2022-07-14:
It is much preferred to have fewer configurations, not a different configuration for every potential variant, whether that consists of a different probe, a different leveling system, or a different [your peripheral here]. If a different UUID is needed for the two types of bed leveling, and this is the primary means of distinguishing these variants, then the UUID may be applied like the other settings that are contingent on the bed leveling system. If there are other settings that need to differ based on the form of bed leveling, those may also be applied in your next PR.

Unfortunately your PRs are not editable by admins. This could be due to a glitch in GitHub, but double-check the "allow edits by maintainers" checkbox in your next submission.

sgparry, 2022-07-16:
I think it has something to do with the fact that the repo is within an organisation rather than being a personal one - do I have to invite you as a collaborator? The documentation is a s clear as mud on this...

thisiskeithb, 2022-07-25:
Since both https://github.com/MarlinFirmware/Configurations/pull/772 & https://github.com/MarlinFirmware/Configurations/pull/773 have been opened to address the reported issue, I'm closing this.

@thinkyhead will need to decide which one to merge.

closed 2022-07-25 by thisiskeithb

sjasonsmith, 2024-04-20:
FYI, I have closed both of these PRs without merging them. This was more of a housekeeping operation than anything, to eliminate old PRs that were going nowhere. If the change still feels important a new PR would be welcome with a single proposed solution, and with the "Allow edits from maintainers" checkbox selected when posting the pull request.