MarlinFirmware/Configurations #803: Error: missing binary operator ?

Ergonomicmike, 2022-08-20:
I know nothing about coding. So, unfortunately, I am trying to hack my way thru a build of Jyers Marlin Bugfix.

I am using the latest nightly custom configuration for the Ender 3 V2, creality UI.

I receive this error:

c:\marlin\configuration.h:553:27: error: missing binary operator before token "("
  553 |  #if TEMP_SENSOR_IS_MAX_TC(0)
      |                           ^
c:\marlin\configuration.h:557:26: error: missing binary operator before token "("
  557 | #if TEMP_SENSOR_IS_MAX_TC(1)

When I comment out these lines  below, the build completes. (Well, I still have some Sanity issues with configuration.h. But I move past this first error.)

```
// Resistor values when using MAX31865 sensors (-5) on TEMP_SENSOR_0 / 1
#if TEMP_SENSOR_IS_MAX_TC(0)
   #define MAX31865_SENSOR_OHMS_0      100 // (Ω) Typically 100 or 1000 (PT100 or PT1000)
   #define MAX31865_CALIBRATION_OHMS_0 430 // (Ω) Typically 430 for Adafruit PT100; 4300 for Adafruit PT1000
#endif

#if TEMP_SENSOR_IS_MAX_TC(1)
   #define MAX31865_SENSOR_OHMS_1      100
   #define MAX31865_CALIBRATION_OHMS_1 430
#endif
```

ellensp, 2022-08-20:
Your using miss matched confgs and code

If using nightly configs https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.1.x.zip
you must use nightly code also. https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip

closed 2022-08-20 by ellensp

Ergonomicmike, 2022-08-20:
Thanks for the quick reply. I'm using a (non-nightly) Marlin-bugfix-JeyersUI. 

I just checked the non-nightly Configuration. The code here is exactly the same.

I don't know enough to know if there's a syntax error in this bit of code.

ellensp, 2022-08-20:
you cannot mix configs from here with 3rd party forks of marlin code either

ellensp, 2022-08-20:
Its not a bug, it a user error
Latest code has the macro in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/core/macros.h#L737

Ergonomicmike, 2022-08-20:
Thanks again. 

FYI, the guys over at the jyers fork link to the Marlin Configuration files. So newb that I am, I thought that those files were acceptable.