MarlinFirmware/Configurations #805: Too many icons for Creality Ender Display board rev1.2

Ergonomicmike, 2022-08-23:
### Bug Description

When - to use the Jyers' UI - I flashed the 'private' file to my Ender 3 V2 Display with a rev 1.2 PCB, the flash fails about 7/8th's into the flash operation.

### Steps to Reproduce

If you have a Display with a rev 1.2 PCB, try it yourself. Copy 'private' from
Configurations/config/examples/Creality/Ender-3 v2/LCD files 
into your SD card, put your card in your Display, power up and wait.

You'll see the icon files being saved one by one, Until it stops.

**Expected behavior:**

Naturally all the icons should be saved and the flash operation should complete successfully.

**Actual behavior:**

The flash operation fails with an "Error."

#### Additional Information

I suspect that Creality 'cheaped out' on memory on this version of the Display board. 

Because I was able to load the icons that I needed by deleting the icon set with a 3 in it, thus reducing the memory footprint.

I also notice that every icon comes in a pair. I don't know if this was by design or a mistake. If the latter, then the obvious solution it to eliminate the duplicate icon, reducing the memory footprint.

ellensp, 2022-08-23:
There are now 3 different types of this display. (that we have seen thus far) https://marlin.crc.id.au/forum/t/identifying-ender-3-v2-screen-versions/960
the 2nd and 3rd are completely undocumented

ellensp, 2022-08-23:
"icon comes in a pair."

There is no such thing.
All icons are unique 

Some are subtitle having a darker or lighter background colour when selected or not.

Ergonomicmike, 2022-08-23:
Attached is a photo of my Rev 1.2 Board, to add to your collection.

It has the F1C100s also. And it also requires "private" to upload files to it.
[image]

On the "pair" of icons: Okay. that makes sense. (It wasn't obvious to me on my monitor.) 

Thanks for the quick reply.

ellensp, 2022-08-23:
It is unlikely to be the number of icons, more likely to be the total space they take up

try these replacing the icons with these, they have been a bit better packed.

[repacked-icons.zip](https://github.com/MarlinFirmware/Configurations/files/9399648/repacked-icons.zip)

Ergonomicmike, 2022-08-23:
It worked! AND it fixed a display with the display that I thought was due to z-ordering.

(I didn't expect shrinking the files to work, given that hard drives use clusters, and even a 1 bit file uses, say, 4k of memory. But apparently on-chip memory is different and doesn't use clusters.)

Thanks for making the extra effort.

closed 2022-08-23 by Ergonomicmike

Ergonomicmike, 2022-08-24:
You solved the problem by repacking/re-compressing the icons. So this is now a philosophical question:

You had told me "Some are subtle having a darker or lighter background colour for when icon is selected or not."

Are the two "pair" of icons for Touch Screens then? Because I don't notice any changes when I click on them using the scroll wheel on my Display. So I was thinking that maybe one is dimmer than the other to indicate a "touch" by a user?

If yes, then are the "pairs" necessary for non-Touch displays?

Ergonomicmike, 2022-08-24:
Also, FYI, I just overloaded the Private directory. Immediately I received a message that the max Memory is 1 MB.