MarlinFirmware/Configurations #823: ENDSTOP_INTERRRUTS break blouch on creality boards, possibly other STM ones

EvilGremlin, 2022-10-06:
z_min_probe is stuck triggered with interripts, i think it's better to disable them at least for creality boards

thisiskeithb, 2022-10-06:
How many reports have we seen with this issue? Were they using genuine BLTouches or clones?

EvilGremlin, 2022-10-06:
Couple so far, one in dicsord, other in local forum. Clone or not doesnt matter - it fails in force_sw too. Fixing configs should be stopgap until someone get to actual issue

ellensp, 2022-10-06:
From a user that just disabled ENDSTOP_INTERRRUTS and it started to work
"its a ender 3 pro, stock mobo, 4.2.2, 256k i think? though it may be 512k but im flashing with 256k anyways because i got it recently. BLTouch is plugged into the BL Touch port on the mobo"
BLTOUCH looks genuine

thisiskeithb, 2022-10-06:
> it fails in force_sw too

Something is really broken if `BLTOUCH_FORCE_SW_MODE` is now causing BLTouches to fail.

EvilGremlin, 2022-10-06:
??? 
it doesn't

silycr, 2022-10-19:
Using Z_Min Endstop with 3dTouch on Tronxy X5S-2e stock board (Chitu cxy-v5-180409) STM32F1. Works both EN and DISABLED ENDSTOP_INTERRRUPTS. 
EN BLTOUCH_FORCE_SW_MODE can not detect 3dTouch. DISABLED can be sensed. Possible 3DTouch issue. Will test with genuine BLTouch Classic.

bellertm1, 2022-11-19:
Hardware: Ender3 Pro and 5, Official Creality BLTouch and STM32F103RE 4.2.7 board
Software: tried 2.1.X Bugfix,   Release 2.1.1,  and 2.0.9.5 all with 512 Flash 
Settings:
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
AND
// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING
AND
 * The BLTouch probe uses a Hall effect sensor and emulates a servo.
 */
#define BLTOUCH

Have tried every variation with same result - after homing, Bed & Extruder heat up and print time shows in progress through completion and Extruder never moves (EEPROM_CHITCHAT shows "Stopped"). Prints reliably as long as BLTouch is not enabled (same for Ender 3 Pro and Ender 5 Pro with same hardware configurations.