MarlinFirmware/Configurations #825: [BUG] Creality V422 not building on release 2.1.1

alf1e, 2022-10-09:
### Bug Description

Fails to build for creality V422 complaing version 2.1.1 config files are out of date for marlin 2.1.1

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/9741289/Configuration.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Copy and paste marlin config files into marlin 2.1.1
2. enable BLTouch in Configuration.h
3. build under any build option 

**Expected behavior:**

Compile successfully 

**Actual behavior:**

Complains that files are wrong version for this marlin
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

EvilGremlin, 2022-10-09:
because you dounloaded Marlin 2.1.x branch, which is not 2.1.1

alf1e, 2022-10-09:
> because you downloaded Marlin 2.1.x branch, which is not 2.1.1

It says in the auto builder that it's marlin version 2.1.1

EvilGremlin, 2022-10-09:
exact download links?

alf1e, 2022-10-09:
> exact download links?

Not at computer right now so don't have them but when I get back should I try downloading the 2.1.1 zip file?

EvilGremlin, 2022-10-09:
Aactually, it might say 2.1.1 in ABM, even though config version was bumped up. Anyway - error is right, this is not a bug.

alf1e, 2022-10-09:
> Aactually, it might say 2.1.1 in ABM, even though config version was bumped up. Anyway - error is right, this is not a bug.

Oh ok sorry

closed 2022-10-09 by alf1e