MarlinFirmware/Configurations #843: [BUG] Configuration Samples failed for machines with MAX6675 sensor

CoolZeroNL, 2022-11-13:
### Bug Description

Im trying to build the bugfix-2.1.x for my Wanhao Dublicator D6, but its failing on compiling:

```
Marlin/src/module/temperature.cpp:1899:6: warning: #warning "Safety Alert! Disable IGNORE_THERMOCOUPLE_ERRORS for the final build!" [-Wcpp]
     #warning "Safety Alert! Disable IGNORE_THERMOCOUPLE_ERRORS for the final build!"
      ^~~~~~~
Compiling .pio/build/mega2560ext/src/src/module/tool_change.cpp.o
Marlin/src/module/temperature.cpp: In static member function 'static raw_adc_t Temperature::read_max_tc(uint8_t)':
Marlin/src/module/temperature.cpp:3101:7: error: 'TEMP_SENSOR_2_MAX_TC_TMAX' was not declared in this scope
       TEMP_SENSOR_2_MAX_TC_TMAX
       ^
Marlin/src/module/temperature.cpp:3079:48: note: in definition of macro 'THERMO_SEL'
       #define THERMO_SEL(A,B,C) (hindex > 1 ? (C) : hindex == 1 ? (B) : (A))
                                                ^
Marlin/src/module/temperature.cpp:3101:7: note: suggested alternative: 'TEMP_SENSOR_0_MAX_TC_TMAX'
       TEMP_SENSOR_2_MAX_TC_TMAX
       ^
Marlin/src/module/temperature.cpp:3079:48: note: in definition of macro 'THERMO_SEL'
       #define THERMO_SEL(A,B,C) (hindex > 1 ? (C) : hindex == 1 ? (B) : (A))
                                                ^
```

link to complete log: https://github.com/CoolZeroNL/Marlin-Build/actions/runs/3454345603/jobs/5765582463#step:5:313

For some reason its asking/looking for TEMP_SENSOR_2, but only TEMP_SENSOR_0 and TEMP_SENSOR_1 is used.

ps, i also tried the configuration files for the machine: Malyan M180, that also uses the MAX6675, but failed with the same error.

### Configuration Files

`Configuration.h` ==> https://github.com/CoolZeroNL/Marlin-Build/blob/main/configuration/wanhao_dublicator_4s/2.1.1/Configuration.h
`Configuration_adv.h` ==> https://github.com/CoolZeroNL/Marlin-Build/blob/main/configuration/wanhao_dublicator_4s/2.1.1/Configuration_adv.h

ellensp, 2022-11-13:
This is not a configuration bug
This is a real bug in marlin code.

ellensp, 2022-11-13:
tentative fix

```DIFF
--- a/Marlin/src/module/temperature.cpp
+++ b/Marlin/src/module/temperature.cpp
@@ -3073,11 +3073,17 @@ void Temperature::disable_all_heaters() {
     #endif
 
     #if HAS_MULTI_MAX_TC
-      // Needed to return the correct temp when this is called between readings
       static raw_adc_t max_tc_temp_previous[MAX_TC_COUNT] = { 0 };
-      #define THERMO_TEMP(I) max_tc_temp_previous[I]
-      #define THERMO_SEL(A,B,C) (hindex > 1 ? (C) : hindex == 1 ? (B) : (A))
-      #define MAXTC_CS_WRITE(V) do{ switch (hindex) { case 1: WRITE(TEMP_1_CS_PIN, V); break; case 2: WRITE(TEMP_2_CS_PIN, V); break; default: WRITE(TEMP_0_CS_PIN, V); } }while(0)
+      #if MAX_TC_COUNT > 2
+        // Needed to return the correct temp when this is called between readings
+        #define THERMO_TEMP(I) max_tc_temp_previous[I]
+        #define THERMO_SEL(A,B,C) (hindex > 1 ? (C) : hindex == 1 ? (B) : (A))
+        #define MAXTC_CS_WRITE(V) do{ switch (hindex) { case 1: WRITE(TEMP_1_CS_PIN, V); break; case 2: WRITE(TEMP_2_CS_PIN, V); break; default: WRITE(TEMP_0_CS_PIN, V); } }while(0)
+      #elif MAX_TC_COUNT > 1
+        #define THERMO_TEMP(I) max_tc_temp_previous[I]
+        #define THERMO_SEL(A,B,C) ( hindex == 1 ? (B) : (A))
+        #define MAXTC_CS_WRITE(V) do{ switch (hindex) { case 1: WRITE(TEMP_1_CS_PIN, V); break; default: WRITE(TEMP_0_CS_PIN, V); } }while(0)
+      #endif
     #else
       // When we have only 1 max tc, THERMO_SEL will pick the appropriate sensor
       // variable, and MAXTC_*() macros will be hardcoded to the correct CS pin.
```

ellensp, 2022-11-13:
Issue was intruded here https://github.com/MarlinFirmware/Marlin/pull/24898

CoolZeroNL, 2022-11-13:
> tentative fix
> 
> ```diff
> diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
> index 37996a80ea..8742cf36df 100644
> --- a/Marlin/src/module/temperature.cpp
> +++ b/Marlin/src/module/temperature.cpp
> @@ -3076,8 +3076,13 @@ void Temperature::disable_all_heaters() {
>        // Needed to return the correct temp when this is called between readings
>        static raw_adc_t max_tc_temp_previous[MAX_TC_COUNT] = { 0 };
>        #define THERMO_TEMP(I) max_tc_temp_previous[I]
> -      #define THERMO_SEL(A,B,C) (hindex > 1 ? (C) : hindex == 1 ? (B) : (A))
> -      #define MAXTC_CS_WRITE(V) do{ switch (hindex) { case 1: WRITE(TEMP_1_CS_PIN, V); break; case 2: WRITE(TEMP_2_CS_PIN, V); break; default: WRITE(TEMP_0_CS_PIN, V); } }while(0)
> +      #if MAX_TC_COUNT > 2
> +        #define THERMO_SEL(A,B,C) (hindex > 1 ? (C) : hindex == 1 ? (B) : (A))
> +        #define MAXTC_CS_WRITE(V) do{ switch (hindex) { case 1: WRITE(TEMP_1_CS_PIN, V); break; case 2: WRITE(TEMP_2_CS_PIN, V); break; default: WRITE(TEMP_0_CS_PIN, V); } }while(0)
> +      #elif MAX_TC_COUNT > 1
> +        #define THERMO_SEL(A,B,C) ( hindex == 1 ? (B) : (A))
> +        #define MAXTC_CS_WRITE(V) do{ switch (hindex) { case 1: WRITE(TEMP_1_CS_PIN, V); break; default: WRITE(TEMP_0_CS_PIN, V); } }while(0)
> +      #endif
>      #else
>        // When we have only 1 max tc, THERMO_SEL will pick the appropriate sensor
>        // variable, and MAXTC_*() macros will be hardcoded to the correct CS pin.
> ```

this is working, thanx for the quick fix. Is this been pushed to the bugfix-2.1.x ?

ellensp, 2022-11-13:
now that you have confirmed it works I will create a PR.  I could see how the code was meant to work, but had no way to test it.

ellensp, 2022-11-13:
pr created, closing this issue

closed 2022-11-13 by ellensp