MarlinFirmware/Configurations #865: T_Rex_2+ INVERT_E0_DIR set to true

ghrrum, 2022-12-14:
#define INVERT_E0_DIR set to true

As there is no way to plug in the extruder backwards this is a problem.
Not sure what the protocol is for updating configs on here and I'm hesitant to do so as I'm not much for programming, just reading.

Regardless, testing the fix on it now to ensure function.

_Originally posted by @ghrrum in https://github.com/MarlinFirmware/Marlin/issues/25094#issuecomment-1351711771_

ghrrum, 2022-12-14:
Continuing the problems found over in the one I referenced this from:
https://github.com/MarlinFirmware/Marlin/issues/25094

Changing PID settings and BANG_MAX  to 255 did not resolve the issue, have reset BANG_MAX back to 180. Leaving adjusted PID values.

thisiskeithb, 2022-12-14:
Tagging @InsanityAutomation since they have a T-Rex 2+ and can chime in 🙂

ghrrum, 2022-12-14:
> Tagging @InsanityAutomation since they have a T-Rex 2+ and can chime in 🙂

Thanks for the efforts man, I'm going to keep picking away at the issue until it works. Going to swap out the hotend and see if I ended up having a hardware failure on it.

InsanityAutomation, 2022-12-14:
In the meantime, this should be usable - 

https://github.com/InsanityAutomation/Marlin/tree/Trex_2.0.x_Devel

One thing to consider, across the life of the machine it was shipped with 3 different stepper driver configurations. Confirm which drivers you have in each slot. Note the toggles to select for each at the very top of configuration.h

ghrrum, 2022-12-14:
Brand new hotend, same issue; hotend quits heating after the first layer is done and the fan engages.
Steppers working fine now though @InsanityAutomation, movement issue was related to another bug that got resolved.

ghrrum, 2022-12-14:
I'm rewinding firmware back to OEM and testing to make sure that works and that it's not hardware related.

ghrrum, 2022-12-14:
Can confirm that hotend issue does not exist on the OEM firmware version.

thisiskeithb, 2022-12-14:
> Can confirm that hotend issue does not exist on the OEM firmware version.

Disabling thermal protections will obviously allow it to work without halting.

ghrrum, 2022-12-14:
> > Can confirm that hotend issue does not exist on the OEM firmware version.
> 
> Disabling thermal protections will obviously allow it to work without halting.

Yes, however I did the same in the current bugfix-2.1.x of Marlin and had the same issue with hotend cutting out after fan engagement.

InsanityAutomation, 2022-12-14:
The fan has a split where the airflow hits the heatbreak and part together when engaged. Because of this, extremely aggressive P values are needed on the PID tune. The autotune doesnt handle it well. Hotend should never be heated without the fan at 50% minimum otherwise heat will creep into the cold side. Also make sure the lower segment has not warped to direct the air onto the heater block instead of the nozzle. This will happen easily if printing ABS etc with the bed over 110c.

ellensp, 2022-12-14:
reading though insanity config 
E changes direction depending on what stepper drivers you have

Ie if you have tmc2208's it needs inverted
If you have anything else It doesn't

so changing  INVERT_E0_DIR for you will break it for everyone who has a4988's

InsanityAutomation, 2022-12-14:
Correct, Gen 1 was 4988, Gen 2 was 8825, Gen 3 2208. Isnt it fun catching midstream changes?

ghrrum, 2022-12-15:
Y'all are awesome and I appreciate the continued efforts. 

> Correct, Gen 1 was 4988, Gen 2 was 8825, Gen 3 2208. Isnt it fun catching midstream changes?

That suggests there needs to be a comment for clarification in the TRex config. I'm going to be chewing through @InsanityAutomation 's firmware and see what I can pull to make work.

ghrrum, 2022-12-15:
And I see that is there for the @InsanityAutomation build, good job man. 

It appears much of the issue I'm dealing with is due to how the hotend is behaving based on the PID settings and the fan needing to be engaged during heating. 

I'm tuning in the PID settings now and trying to dial it in and should be able to manage the fan settings from g-code. I'm going to tweak how I have things setup and I'll touch back with results.

Side note, the OEM PID settings are odd.....

ghrrum, 2022-12-15:
I'm still having issues with M106 S255 causing a false thermal runaway

InsanityAutomation, 2022-12-15:
Can you attach youre current config file?

InsanityAutomation, 2022-12-15:
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 60        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 15    // Degrees Celsius

  #define ADAPTIVE_FAN_SLOWING              // Slow part cooling fan if temperature drops
  #if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
    #define NO_FAN_SLOWING_IN_PID_TUNING    // Don't slow fan speed during M303
  #endif

Try this - increasing the hysteresis gives adaptive fan slowing more room to work. This will allow firmware the throttle the fan as the hotend temperature drops to prevent the runaway.

#define BANG_MAX 255     // Limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current

Turn up the max power, may not even be able to hold at that low of a rate

Try these PID values, a bit more aggressive

#define DEFAULT_Kp_LIST {  22.20,  22.20 }
      #define DEFAULT_Ki_LIST {   1.08,   1.08 }
      #define DEFAULT_Kd_LIST { 114.00, 114.00 }

ghrrum, 2022-12-15:
Plugging it in now. Was hesitant to push values that high. 

Lets see if it catches fire.

ghrrum, 2022-12-15:
Dry run looks good, still dipping all the way down to 185C from 200C setpoint on fan engagement, overshooting temp by about 15C too

ghrrum, 2022-12-15:
a 30C variance on the initial swing but looks like it's stabilizing to +/- .5C which does fit well. lets see if it will actually print something now.

InsanityAutomation, 2022-12-15:
Sounds promising! That's not unexpected on a shock with the fan starting, so hopefully we're in good shape!

ghrrum, 2022-12-15:
Temperature swing on the print was only -5C and stabilized at +/-1.5C

I think that might be solved

ghrrum, 2022-12-15:
Temperature swing down to 196.3C with 210C setpoint temperature more or less stable at 204C

ghrrum, 2022-12-15:
Test print successful!
I'm calling this done and functional now.
Thanks for the help guys!

closed 2022-12-15 by ghrrum

ghrrum, 2022-12-15:
Still some tweaking to values on the PID needed.

ghrrum, 2022-12-19:
Upon further discussion and consultation reconfigured physical print head and cooling fan to further restrict air flow.