MarlinFirmware/Configurations #867: [BUG] Marlin 2.1.2 CR-30 "jumpy" motion

ruralguru, 2022-12-22:
### Bug Description
Very slow motion on short runs. Wrote down settings from firmware as shipped, Only 

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.
Changes in config: 
_Bootscreen.h: width 120
* all mods in config and config_Adv are commented with // ERW
=== config:
preheat2: PETG content
Baudrate: 1000000
X & Y _BED_SIZE

=== ADV Config
HOST_ACTION_COMMANDS
CANCEL_OBLECTS
PAUSE_PRINT
HOTEND_IDLE_TIMEOUT
Silent fan stuff
Animated Logo
### Steps to Reproduce

1. Use electrical tape fix for USB back feed from PI.
2. Install Marlin2.1.2 release
3. use Marlin 2.1.2 bug fix config examples as directed in guide.
4. install using SD
5. generate cube (used creality slicer) 15% infill gyroid.
6. print
7. wait for infill or any other short path.

**Expected behavior:**

make smooth motions on short jogs, Display functional.

**Actual behavior:**

sooth motion on long runs, slow jittery motion in short runs, printer stops responding to display and rotary selector, when stopping print it "pauses" and wants to reheat(have not allowed that). also DEFAULT_ACCELERATION when set to 600 stays at 300.

#### Additional Information
* Included image of print, bubbles visble, though it seems fine
*  E0 motion with end of infill and start of next shell.
* Display with lack of response requiring time extensive command
* Build bin file for stock CR-30
* octopi connection output text file.

I have attempted the following:
* increase speed
* Increase Acceleration
* Remove all Host commands
* 0% infill, this was way faster,, but still slow with the X, Y and Z lettering jobs.
* don't connect Pi
* Remove host keep alive
* Use arc-welder plugin with creality slicer

[ERW_CR30_issues.zip](https://github.com/MarlinFirmware/Configurations/files/10290920/ERW_CR30_issues.zip)

ruralguru, 2022-12-23:
Update:
Same issues exists with fresh build of the 2.1.2 release of configurations and firmware. Used fresh spin of Ubuntu 22 with VS code to ensure not contamination of packages.
Returned to stock firmware to confirm its not a hardware issue. prints smoothly and display is responsive.

thisiskeithb, 2024-06-27:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

ruralguru, 2024-09-17:
Just saw this. have a few long prints to get through but it is on my list now. thank you for the update. will report back, likely to be a while.