MarlinFirmware/Configurations #875: Part Cooling Problem

BitschiClaudio, 2022-12-31:
Hi Guys,
I have a problem.

Marlin 2.1.2
BTT octopus pro
2 hot ends 2 extruders
2 Part coller fan

Active:
#define REDUNDANT_PART_COOLING_FAN 1  // Index of the fan to sync with FAN 0.
#define FANMUX0_PIN 2
#define FANMUX1_PIN 2

I want both Part_Fan to always be running, at printing.

If extruder 0 is active in pronterface, everything works with 
M106 S
M106 T0/T1 S0-255
but
with active extruder 1 in Pronterface

nothing works..

Does somebody has any idea?

And a new good year

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/10327117/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/10327118/Configuration_adv.txt)

_Originally posted by @BitschiClaudio in https://github.com/MarlinFirmware/Marlin/discussions/22984#discussioncomment-4559697_

BitschiClaudio, 2022-12-31:
gkljl,

BitschiClaudio, 2022-12-31:
shz

thisiskeithb, 2022-12-31:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/874

closed 2022-12-31 by thisiskeithb