MarlinFirmware/Configurations #900: [BUG] Unable to build Malyan/M200 v1 example configuration according to its README.md

SharkWipf, 2023-03-06:
### Bug Description

Exactly following the instructions from the README.md, without any changes to the example configs, I am unable to finish the building process due to the following error:

```
In file included from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/clock.h:43,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring_time.h:23,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:38,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\shared\Marduino.h:36,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\STM32\HAL.h:28,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\HAL.h:30,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\inc\MarlinConfig.h:31,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\STM32\HAL.cpp:27:
C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\STM32\HAL.cpp: In static member function 'static void MarlinHAL::init()':
C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/stm32_def.h:58:17: error: the value of 'SystemCoreClock' is not usable in a constant expression
   58 |   #define F_CPU SystemCoreClock
      |                 ^~~~~~~~~~~~~~~
C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\STM32\HAL.cpp:70:27: note: in expansion of macro 'F_CPU'
   70 |   constexpr int cpuFreq = F_CPU;
      |                           ^~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h:133,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f1xx.h:131,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/stm32_def.h:28,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/clock.h:43,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring_time.h:23,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:38,
                 from C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\shared\Marduino.h:36,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\STM32\HAL.h:28,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\HAL.h:30,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\inc\MarlinConfig.h:31,
                 from C:\Users\User\Downloads\Marlin-bugfix-2.1.x\Marlin\src\HAL\STM32\HAL.cpp:27:
C:\Users\User\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/system_stm32f1xx.h:51:17: note: 'uint32_t SystemCoreClock' is not const
   51 | extern uint32_t SystemCoreClock;          /*!< System Clock Frequency (Core Clock) */
      |                 ^~~~~~~~~~~~~~~
```

This error happens on both the suggested (now deprecated) STM32 Cores versions 1.8 and 1.9 (1.7 fails on a different error) and on [the latest STM32 2.4.0](https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/main/package_stmicroelectronics_index.json).

I have tested both the 2.1 stable and 2.1 bugfix/nightly branches/configs.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
Unmodified configs straight from the repo, repacked into a zip:
[Marlin.zip](https://github.com/MarlinFirmware/Configurations/files/10894030/Marlin.zip)

### Steps to Reproduce

Simply follow the installation instructions from the Maylan/M200 v1 README.md.

**Expected behavior:**
A completed build.

**Actual behavior:**
No completed build.

#### Additional Information
Full build log: 
[marlinfull.log](https://github.com/MarlinFirmware/Configurations/files/10894065/marlinfull.log)

ellensp, 2023-03-06:
add `#define F_CPU 72000000`  to Your configuration.h file.  
This gets past the error, but im only guessing its running the STM32F103CB MPU at 72 MHz

SharkWipf, 2023-03-06:
Thanks, I'll give that a go and report back. Is figuring out the correct clockspeed a matter of trial and error, or can I otherwise verify it?
Is this actually affecting anything important? As apparently it wasn't necessary to specify in the past?

ellensp, 2023-03-06:
It something broken in arduino ide..  works fine in platformio, and breaks if you add the define. 

so currently you cannot win. 
You cant add it as that breaks platformio
you cant not add it as that breaks arduino ide...

SharkWipf, 2023-03-06:
Well, the firmware installs, and beyond that I have no idea. It doesn't show up over serial and it doesn't seem to connect to the LCD over serial either. Could this be related to the CPU frequency being wrong? I have no idea what it is used for.  
Interestingly, if I change the board version to v2 (despite the board being a v1 pretty sure, no back fan), the sensors do start working, briefly, before bootlooping.  
Either way it seems like there's more not (longer) accurate about this example config, at least for the v1.

I tried both the instructions from the README.md and the ones from @xC0000005's original repo, both with just the unmodified (except for the `F_CPU` line) config files from this repo, and neither seem to produce a working build for the v1 (anymore).

xC0000005, 2023-03-06:
If you are trying to flash a V1 and not getting USB serial, be absolutely certain the board overrides to toggle that pull-up 
are present, t’s required to signal USB bus enumeration on the V1.