MarlinFirmware/Configurations #901: [BUG] Issues with the Ender 5 S1 Configuration (Won't Build)

jrgleason, 2023-03-07:
### Bug Description

I am trying to build with the configuration provided by the Ender 5 S1 folder. However, when I do I get the following...

`Unknown MOTHERBOARD value set in Configuration.h.`

It appears the `BOARD_CREALITY_F401RE` board is not added by default and I am not sure how to add it

### Configuration Files

No modification see [this](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-5%20S1)

### Steps to Reproduce

1. Clone either the BTT or Vanilla version of Marlin
2. Replace the configuration files with the ones provided
3. Run `pio run`
4. Note the errors

**Expected behavior:**

I would like to be able to compile the bin file to deploy to the printer.

**Actual behavior:**

I get multiple errors...

```
Processing mega2560 (board: megaatmega2560; platform: atmelavr@~3.4; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3145:15: error: missing binary operator before token "("
 #if DGUS_UI_IS(MKS)
               ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3147:17: error: missing binary operator before token "("
 #elif DGUS_UI_IS(IA_CREALITY)
                 ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34:0,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:886:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h."
     #error "Unknown MOTHERBOARD value set in Configuration.h."
      ^~~~~
Error: Failed to parse Marlin features. See previous error messages.
========================== [FAILED] Took 1.14 seconds ==========================
```

thisiskeithb, 2023-03-07:
Ender-5 S1 support is still pending in the main Marlin repo: https://github.com/MarlinFirmware/Marlin/pull/25382

> Processing mega2560 (board: megaatmega2560; platform: atmelavr@~3.4; framework: arduino)

You're also building with the wrong environment. You'll build with `STM32F401RE_CREALITY` using the referenced PR above.

jrgleason, 2023-04-30:
Can you possibly be a little clearer with how I run under that environment. I also noticed there is a 2.0.x repo from creality available here https://github.com/CrealityOfficial/Ender-5S1