MarlinFirmware/Configurations #906: [BUG] 3x3 Bilinear Auto Bed Leveling Breaks Everything

viniciusbrit, 2023-03-14:
1. I've tested in both release and bugfixes.

### Bug description:
I'm using an Aus3d IR Bed Probe, on an SKR MINI E3 V2.0, Regular Ender 3.
When configuring, I enabled 3x3 Bilinear Grid ABL. Then 2 days of pain came.
First, my printer stopped compensating for the dead space near the initial Y coordinates, then the sensor, when Bed leveling, would crash the routine right after reading one of the grid points, this wasn't consistent, but it happened mostly on the 3rd, 4th and 6th points.
There wasn't any IR interference, as I was in a pitch black room.
Some settings, like fade height, Z probe offset, software endstops and some others, did not apply changes after a firmware update (while I was attempting to fix it) or just didn't work at all.
I left some comments detailing this on the specific parts that broke.
I should note that I only tested 2x2, 3x3 and 4x4 grids. The only one that turned out like this was the 3x3 grid.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/10974008/Configuration.zip)
There are comments in these detailing everything.

### Steps to Reproduce

1. Change the settings to what I set.
2. Set bed leveling on.
3. Wait to see what you get.

**Expected behavior:**

3x3 Bilinear grid bed leveling success, inside the constraints of the bed, without anything weird.

**Actual behavior:**

Either a crash of the Auto bed leveling process, returning you to the home screen.
Or settings not applying even after the leveling process.

thisiskeithb, 2023-03-14:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

thisiskeithb, 2023-03-14:
Actually, you should open a proper bug report in the main repository since this repo is for reporting bugs with our hosted configs: https://github.com/MarlinFirmware/Marlin/issues/new/choose

closed 2023-03-14 by thisiskeithb