MarlinFirmware/Configurations #908: In JG Aurora A1 and A5S advanced configs,  FAN_ACTIVE_STATE LOW should be FAN_INVERTING 1

CFTBL, 2023-03-17:
The example Configuration_adv.h for JG Aurora A1 and A5S contains the following line:

#define FAN_ACTIVE_STATE LOW

FAN_ACTIVE_STATE isn't implemented. It must be replaced with:

#define FAN_INVERTING 1

...otherwise the part cooling fan will react opposite the way it should (for example, M106 will turn it off, M107 will turn it on).

This error showed up in the JG Aurora A1/A5S example configuration for release 2.0.9.3 and all releases since then, up to and including the current 2.1.x bugfix examples.

It appears in the 2.0.9.4 Example Configuration README, under Changes,  as...

FAN_INVERTING => FAN_ACTIVE_STATE (LOW)

CFTBL, 2023-03-18:
I went ahead and submitted a pull request, #909, to fix this.

thisiskeithb, 2023-03-18:
Closing since you've opened a PR.

closed 2023-03-18 by thisiskeithb