MarlinFirmware/Configurations #921: [BUG] 2.1.2 Release build fails using Anycubic Chiron example configs

SwiftNick, 2023-04-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using the example config files for the AnyCubic Chiron, compilation fails due to missing LCD_SERIAL_PORT definition.

The previous definition in configuration.h has been removed  [2.1.1 config file](https://github.com/MarlinFirmware/Configurations/blob/f52c85b0e3b241f8fabb4a902a4aa53775fdeb93/config/examples/AnyCubic/Chiron/Configuration.h#L2990)
There is some refactoring around the DGUS_LCD code which seems to relate to this.

In configuration.h, restoring LCD_SERIAL_PORT 3 or adding HAS_DGUS_LCD 1 to the code block resolves the issue.

```
#define ANYCUBIC_LCD_CHIRON
#if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
  #define HAS_DGUS_LCD 1
```  
  @thinkyhead  not sure which you would prefer as I don't know the extent of your refactoring.
  

### Bug Timeline

Introduced in 2.1.2 release

### Expected behavior

Code should build without error

### Actual behavior

Compilation fails with sanity check error

### Version of Marlin Firmware

2.1.2

### Printer model

Anycubic Chiron

### Electronics

stock build

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-04-15:
to me the thing to do is this, in code...

```DiFF
diff --git a/Marlin/src/inc/Conditionals_LCD.h b/Marlin/src/inc/Conditionals_LCD.h
index 60bc653560..cfb72b6745 100644
--- a/Marlin/src/inc/Conditionals_LCD.h
+++ b/Marlin/src/inc/Conditionals_LCD.h
@@ -477,7 +477,7 @@
 #endif
 
 // Aliases for LCD features
-#if !DGUS_UI_IS(NONE) || ENABLED(ANYCUBIC_LCD_VYPER)
+#if !DGUS_UI_IS(NONE) || ENABLED(ANYCUBIC_LCD_VYPER) || ENABLED(ANYCUBIC_LCD_CHIRON)
   #define HAS_DGUS_LCD 1
   #if DGUS_UI_IS(ORIGIN, FYSETC, HIPRECY, MKS)
     #define HAS_DGUS_LCD_CLASSIC 1
```

SwiftNick, 2023-04-15:
> > I should set it in the example configuration, since the Chiron's LCD can be hooked up to any serial port on any board
> 
> Transferring to the Configurations repo…

I haven't submitted a PR as Scott mention the LCD code was being refactored so I wasn't sure of the best place to fix this.

thisiskeithb, 2023-04-24:
Fixed in https://github.com/MarlinFirmware/Configurations/pull/927

closed 2023-04-24 by thisiskeithb

thisiskeithb, 2023-04-25:
My PR was not merged since this was patched upstream in Marlin: https://github.com/MarlinFirmware/Marlin/commit/a2c0cc0e7b0fcbeddf95995d916ff0a23c6f1c37

I'm sure there will be a follow-up since CI is failing.

SwiftNick, 2024-02-10:
2.1.2.2 release also fails to build with the same issue missing LCD_SERIAL_PORT  definition.

Bugfix 2.1.x branch builds fine.

ellensp, 2024-02-10:
"Marlin 2.1.2.2 is a maintenance release with several patches to improve existing features, with minor adjustments to configuration." 

It is not a fully updated marlin, you need to wait for 2.1.3