MarlinFirmware/Configurations #935: EasyThreed K7 ET4000+ AT32F403ARCT7 CHIP cant print on marlin 2.1.2.1 alway hang when start print

flishy2532, 2023-05-19:
EasyThreed K7 ET4000+ AT32F403ARCT7 CHIP cant print on marlin 2.1.2.1 alway hang when start print
Any one can help ?
am update firmware easy threed k7 et4000+ to mainboard AT32F403ARCT7 CHIP 
when start print it alway hang on center printer head not move 
please replay me
thank you
best reguard

thisiskeithb, 2023-05-19:
> am update firmware easy threed k7 et4000+ to mainboard AT32F403ARCT7 CHIP

The default config for this machine is for a `MKS_ROBIN_LITE` which has a `STM32F103RC`, not `AT32F403ARCT7`:

https://github.com/MarlinFirmware/Configurations/blob/d39182cd099df759d8207e3c35db489a6e705554/config/examples/EasyThreeD/ET4000PLUS/Configuration.h#L72

Until AT32 support is added, you'll have to use the vendor-provided copy of Marlin.

EvilGremlin, 2023-05-19:
Oh FFS, not fourth one! 
*insert MichaelOhNoGodNo.gif*

flishy2532, 2023-05-19:
> The default config for this machine is for a `MKS_ROBIN_LITE` which has a `STM32F103RC`, not `AT32F403ARCT7`: [...]

ah when add AT32F403ARCT7

thisiskeithb, 2023-05-19:
> not fourth one!

This came up once before: https://github.com/MarlinFirmware/Marlin/issues/24894

> ah when add AT32F403ARCT7

Whenever someone submits a PR to add support.

EvilGremlin, 2023-05-19:
Not in any forseeable future, maybe never. It has to be arduino/platformio framework, and not a patched copypaste of STM code. Some forks could pop up, maybe even good working ones, but code/support has to be of certain qualities (and legality) to get into vanilla.

flishy2532, 2023-05-20:
any one can help build easythreed k7 et4000+ (genericAT32F403ARCT7 chip)  ?

[image]

 ?