MarlinFirmware/Configurations #948: Creality Ender 5 S1 configuration files Compile Error

kipp6912, 2023-06-11:
Creality Ender 5 S1 configuration files Compile Error

Will not compile with Auto Build Marlin with Creality Ender 5 S1 configuration files

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Marlin-Configfiles.zip](https://github.com/MarlinFirmware/Configurations/files/11713737/Marlin-Configfiles.zip)

If you've made any other modifications describe them in detail here.
No modifications were made
### Steps to Reproduce

1. Compile bugfix-2.1.x with Auto Build Marlin no errors
2. Add configuration Configuration_adv.h and Configuration.h for Creality Ender 5 s1 into marlin folder overwriting the originals and recompile with Auto Build Marlin will not compile error given is -> Marlin\src\gcode\ota\M936.cpp:30:10: fatal error: ../../lcd/rts/lcd_rts.h: No such file or directory 30 | #include "../../lcd/rts/lcd_rts.h"  |   ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated. *** [.pio\build\STM32F401RE_creality\src\src\gcode\ota\M936.cpp.o] Error 1
3. [and so on...]

**Expected behavior:**

I expected it to compile with no errors.

**Actual behavior:**

Compile Error -> Marlin\src\gcode\ota\M936.cpp:30:10: fatal error: ../../lcd/rts/lcd_rts.h: No such file or directory 30 | #include "../../lcd/rts/lcd_rts.h"  |   ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated. *** [.pio\build\STM32F401RE_creality\src\src\gcode\ota\M936.cpp.o] Error 1

#### Additional Information
Image uploaded

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.
[image]

thisiskeithb, 2023-06-11:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/901

thisiskeithb, 2023-06-11:
Ender-5 S1 support is still pending in the main Marlin repo: https://github.com/MarlinFirmware/Marlin/pull/25382

closed 2023-06-11 by thisiskeithb