MarlinFirmware/Configurations #952: Creality CR10s v1 bug: Marlin\src\module\temperature.cpp:330:38: error: static assertion failed: PREHEAT_2_TEMP_HOTEND (250) must be less than HEATER_0_MAXTEMP (250) - 15.

isidorosss, 2023-06-18:


ellensp, 2023-06-18:
Not a bug.

read the  message

ellensp, 2023-06-18:
unless your talking abut the marlin provided config...  
checking

ellensp, 2023-06-18:
This is in example Configs.  

https://github.com/MarlinFirmware/Configurations/blob/release-2.1.2.1/config/examples/Creality/CR-10S/CrealityV1/Configuration.h#L2227

isidorosss, 2023-06-18:
> This is in example Configs.
> 
> https://github.com/MarlinFirmware/Configurations/blob/release-2.1.2.1/config/examples/Creality/CR-10S/CrealityV1/Configuration.h#L2227
> 
> #define HEATER_0_MAXTEMP 250 with #define PREHEAT_2_TEMP_HOTEND 250 is not valid

Thank you for the help. Ok, i'm replacing
#define PREHEAT_2_TEMP_HOTEND 250
with
#define HEATER_0_MAXTEMP 250
right?

I'm updating my Creality CR10s to every newer version of Marlin and it's the first time i get this error on the newest version 2.1.2.1.
Will it be fixed on newer releases?

One more thing. I guess i shouldn't open a new issue for this. On Configuration_adv.h , line 1743, i have to enable #define NO_SD_DETECT . I guess this must be global on CR10s and should be fixed on newer releases. Could it be that only my CR10s has an SD extender?

Thanks!

ellensp, 2023-06-18:
no...  Im just documenting the cause 
with #define HEATER_0_MAXTEMP 250 the highest PREHEAT_2_TEMP_HOTEND you can set is 235

isidorosss, 2023-06-18:
ok thanks. i'll wait for a fix then.

isidorosss, 2023-07-27:
Anyone? Any suggestions?

thisiskeithb, 2023-07-27:
> Anyone? Any suggestions?

Change `PREHEAT_2_TEMP_HOTEND` from `250` to `235`.

thisiskeithb, 2023-07-27:
This is already fixed in `bugfix-2.1.x` and will be fixed in the next tagged release:

https://github.com/MarlinFirmware/Configurations/blob/b2667913e78ac78bf3ada855fe3b2f6cbfe7cf64/config/examples/Creality/CR-10S/CrealityV1/Configuration.h#L2372

closed 2023-07-27 by thisiskeithb