MarlinFirmware/Configurations #970: [BUG] Ender5 plus with new Dwin Set

leseaw, 2023-08-31:


leseaw, 2023-08-31:
### Bug Description

First LCD flashed, also worked end line came.
New design is also on it

After that Marlin flashed, no function, I can press LCD and new windows go up but nothing moves, no temperature control het, nothing happens

### Configuration Files

`Configuration.h` and `Configuration_adv.h`.

from config Files Configurations-release-2.1.2.1.zip and Configurations-release-2.1.2.1

### Steps to Reproduce

1. Update LCD
2. Compile Firmware with VS Code
3. Upload .hex File with Prusa Sclicer
4. Start Printer, nothing is moving or heating

**Expected behavior:**

Printer prints

**Actual behavior:**

Printer only show LCD Dektop, Can select everything but then nothing happens

ellensp, 2023-08-31:
What dwin files did you use?

ellensp, 2023-08-31:
exactly...

the readme https://github.com/MarlinFirmware/Configurations/blob/release-2.1.2.1/config/examples/Creality/Ender-5%20Plus/CrealityV1/README.md say to use https://github.com/coldtobi/Marlin_DGUS_Resources

a completely different screen User interface

the default config is for #define DGUS_LCD_UI_ORIGIN with the linked dwin files

you have installed DGUS_LCD_UI_RELOADED dwin files

Use correct config file setting with the dwin set you installed.

leseaw, 2023-08-31:
why closed?
i test both
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-5%20Plus
with this also not working

ellensp, 2023-08-31:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-5%20Plus also uses #define DGUS_LCD_UI ORIGIN  not reloaded

If you wish to use reloaded dwin set you MUST update the define to
 #define DGUS_LCD_UI RELOADED

leseaw, 2023-08-31:
it says so
and it does not work

`#define DGUS_LCD_UI ORIGIN
#if DGUS_UI_IS(MKS)
  #define USE_MKS_GREEN_UI
#elif DGUS_UI_IS(IA_CREALITY)
  //#define LCD_SCREEN_ROTATE 90          // Portrait Mode or 800x480 displays
  //#define IA_CREALITY_BOOT_DELAY 1500   // (ms)
#endif`

ellensp, 2023-08-31:
Note the syntax has changed from 2,1,2.1 to bugfix

2.1.2.1
```
#define DGUS_LCD_UI_ORIGIN
or
#define DGUS_LCD_UI_RELOADED

```
bugfix 2.1.x
```
#define DGUS_LCD_UI ORIGIN
or 
#define DGUS_LCD_UI RELOADED

```

leseaw, 2023-08-31:
i have an ender 5 plus with bugfix 2.1.x
then I am correct that the template here is correct or not?
And if this template does not work with an unmodified original printer it is a mistake, isn't it?

leseaw, 2023-08-31:
open this please again or i will make a new bug
settings from config with dwin set will not work

ellensp, 2023-08-31:
The issue is that someone has updated the read me incorrectly..

In https://github.com/MarlinFirmware/Configurations/pull/904 
they changed  the read me to reloaded UI but the config file is still for origin UI

leseaw, 2023-08-31:
you cant set reloaded in marlin, you have then to change a lot :(
What has to be done now that the crap runs

ellensp, 2023-08-31:
To use RELOADED UI you need to make these 6 changes/

In Configuration.h change these  3 lines
 
```DIFF
diff config/examples/Creality/Ender-5\ Plus/CrealityV1\ \(ORIGIN\ UI\)/Configuration.h config/examples/Creality/Ender-5\ Plus/CrealityV1\ \(RELOADED\ UI\)/Configuration.h 
2141c2141
<   #define GRID_MAX_POINTS_X 7       // Don't use more than 15 points per axis, implementation limited.
---
>   #define GRID_MAX_POINTS_X 5       // Don't use more than 15 points per axis, implementation limited.
2203c2203
< //#define LCD_BED_TRAMMING
---
> #define LCD_BED_TRAMMING
3177c3177
< #define DGUS_LCD_UI ORIGIN
---
> #define DGUS_LCD_UI RELOADED

```

In Configuration_adv.h change these 3 lines

```DIFF
diff config/examples/Creality/Ender-5\ Plus/CrealityV1\ \(ORIGIN\ UI\)/Configuration_adv.h config/examples/Creality/Ender-5\ Plus/CrealityV1\ \(RELOADED\ UI\)/Configuration_adv.h 
2225c2225
< //#define BABYSTEPPING
---
> #define BABYSTEPPING
2229c2229
<   //#define BABYSTEP_ALWAYS_AVAILABLE       // Allow babystepping at all times (not just during movement)
---
>   #define BABYSTEP_ALWAYS_AVAILABLE         // Allow babystepping at all times (not just during movement)
2248c2248
<   //#define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping
---
>   #define BABYSTEP_ZPROBE_OFFSET            // Combine M851 Z and Babystepping
```

leseaw, 2023-09-01:
now, its working fine
and i can print with ender5 plus :)

EDOL12, 2023-09-04:
Bsr,

je suis surpris de recevoir ce message car j'ai une corexy mais pas de chez Créality.

j'ai une sapphire plus de chez twos trees qui n'est pas un cadeau a cause des différentes versions.

Elle fonctionnait parfaitement mais j'ai brûlé la carte et je réinstalle une MKS Robin Nano V3.1 a la place de la Robin Nano v1.2.

j'ai toujours quelques soucis depuis les cartes 32 bits qui ne se traite plus avec Arduino IDE mais avec VS Code que je n'aime pas et qui me le rend bien.

Si vous avez quelques compétences en la matière, je suis preneur.

Bonne soirée

thisiskeithb, 2023-09-04:
@EDOL12 : This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server