MarlinFirmware/Configurations #975: Creality Ender 5 Plus with 4.2.2 board and ATMEGA2560 chip requires firmware.hex firmware

Dednam, 2023-10-03:
### Bug Description

Hi. I have a quite new Creality Ender 5 Plus with 4.2.2 board and ATMEGA2560 chip. The standard firmware from Creality's website is .hex and new firmware therefore is also requires to be firmware.hex . Marlin unfortunately doesn't allow for this motherboard / chip combination at present and only allows for Architecture STM32F1 and compiling to firmware.bin . I tested in 2.0x, 2.1.x and bugfix-2.1.x

Configuration.h compiles to bin with the best settings as far as I can see. The main settings in the Configuration.h: 
Motherboard:      	MOTHERBOARD BOARD_CREALITY_V422
SERIAL_PORT: 		1
Driver type:          	MC2208_STANDALONE
LCD_SERIAL_PORT:	3

### Configuration Files
[Configuration.zip](https://github.com/MarlinFirmware/Configurations/files/12795648/Configuration.zip)

I also tested with a wrong motherboard  to compile to .hex:
Motherboard:      	BOARD_RAMPS_CREALITY
SERIAL_PORT: 		0
Driver type:          	MC2208
LCD_SERIAL_PORT:	2

### Steps to Reproduce

1. Copy in supplied 2/1/x configuration files
2. Auto Build Marlin
3. Notice only options for compiling to .bin and none to the required .hex.

**Expected behavior:**

Option to compile to .hex for instance with mega2560

**Actual behavior:**

Only options to compile to .bin with STM32F1

#### Additional Information

EvilGremlin, 2023-10-03:
Nothing is wrong. 4.2.2 board and ATMEGA2560 are mutually exclusive. mega2560 is v2 board, which is BOARD_RAMPS_CREALITY

thisiskeithb, 2023-10-03:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

Dednam, 2023-10-03:
> Nothing is wrong. v4.x.x board and ATMEGA2560 are mutually exclusive. mega2560 is v2 board, which is BOARD_RAMPS_CREALITY

Well. My printer has a v4.x.x board and ATMEGA2560 chip. I found that others have the same combination by Googling a bit.

EvilGremlin, 2023-10-03:
photo?

Dednam, 2023-10-03:
Photos of board and chip in the printer:
[image]
Photos of separate new silent board not fitted yet:
[image]

EvilGremlin, 2023-10-03:
Which clearly says Creality v2.2

Dednam, 2023-10-03:
In Printer Creality V.2.2
[image]

Dednam, 2023-10-03:
With chip
[image]

Dednam, 2023-10-03:
On separate new unused silent board: 
[image]

Dednam, 2023-10-03:
Also with ATMEGA2560 chip
[image]

Dednam, 2023-10-03:
That Marlin doesn't allow for at present

EvilGremlin, 2023-10-03:
Again, it's V2 voard which os BOARD_RAMPS_CREALITY. Not 4.2.2