MarlinFirmware/Configurations #979: [BUG] JGAURORA MAGIC unloads filament at print start

nvdias0, 2023-10-12:
After updating to 2.1.2.1 from stock firmware, the filament always auto-unload at print start.

thisiskeithb, 2023-10-12:
The [stock JGAurora Magic config matches](https://github.com/JGMaker3dofficial/magic/blob/cf104ef6cbe44f9e12dc0b17975241ff191c9c9f/magic/Marlin/Configuration.h#L930C2-L936) what we have:

### JGAurora's Magic source
```cpp
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define NUM_RUNOUT_SENSORS   1     // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_INVERTING true // set to true to invert the logic of the sensor.
  #define FIL_RUNOUT_PULLUP          // Use internal pullup for filament runout pins.
```

### MarlinFirmware source
https://github.com/MarlinFirmware/Configurations/blob/0e94d8886d8707e25f1c540739d68c72b1de1948/config/examples/JGAurora/Magic/Configuration.h#L1880-L1885

(`FIL_RUNOUT_INVERTING true` is now `FIL_RUNOUT_STATE HIGH`, so they're the same)

If you send `M119` via serial/USB, what does the `filament:` line say with filament present? What about without filament?

`M119` example:
```
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: filament: TRIGGERED
```

thisiskeithb, 2023-10-12:
If you revert back to the original firmware, what does `M119` report? All of the firmware configs/builds I'm finding for the the JGAurora Magic match what we have and with your machine showing `filament: open` when filament is present & removed, points to a possible physical switch/wiring issue.

nvdias0, 2023-10-13:
I've just used a multimeter to test the wiring and filament detection switch.
I confirm the wiring is working ok up to the mainboard.

Searching in the code, it assumes that **FIL_RUNOUT_PIN** for these mainboards **BOARD_RAMPS_14_EFB** is **SERVO3_PIN** (**pin** 4).

BUT ... Looking at the mainboard plug, the filament runout pin in the mainboard is **X+ / Pin 2** and not SERVO3_PIN
(i've seen the same in several youtube tech tutorials in youtube that shows the mainboard)

So I've added :
**#define FIL_RUNOUT_PIN    X_MAX_PIN**
into configuration.h

And **it worked !**
M119 reports filament open or tiggered as expected !

I'm using a JGMAKER / JGAURORA MAGIC.

Thanks

ellensp, 2023-10-13:
@nvdias0 confirmed they pulled a pin switch on us

https://github.com/JGMaker3dofficial/magic/blob/cf104ef6cbe44f9e12dc0b17975241ff191c9c9f/magic/Marlin/pins_RAMPS.h#L282

  #define FIL_RUNOUT_PIN    2

vs a stock ramps of 4

ellensp, 2023-10-13:
Added to the JGAurora Magic Fixes PR https://github.com/MarlinFirmware/Configurations/pull/980

closed 2023-10-13 by ellensp