MarlinFirmware/Configurations #996: Temperature not stabilizing on Anycubic Predator

GruenSein, 2023-12-24:
### Bug Description

Hi all,

in case anybody reads this: Happy holidays, everyone!
So, here is my issue. I have upgraded my Anycubic Predator from the manufacturers stock firmware to the current Marlin v2.1.2.1 release using the example configuration for the Predator. While building and flashing went well, my machine won't print anything in the current state because it is stuck in the heating phase of the extruder. I have waited for almost 20 minutes watching the temperatures fluctuating around the setpoint of 205C. It never settles.

Things, I have already tried that did not solve the problem:

- Running the PID auto-tuning
- I noticed that the fans do not turn on automatically. I tried switching them on manually to "slow the heating" and prevent sudden temperature spikes but this did not help the issue.
- Going back to Marlin v2.0.x and the corresponding configuration
- Adapting the PID_K1 to 0.75 or 0.5 and increasing the PID_FUNCTIONAL_RANGE to 50 as suggested here](https://github.com/MarlinFirmware/Marlin/issues/3211). Then rerunning the PID autotune

As my machine used to work reasonably well with the manufacturer's firmware, I assume that this must be a configuration issue but I am out of ideas. Therefore, I'd appreciate any help that anyone can provide.

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Configurations/files/13761260/config.zip)

Modifications as mentioned above:
- PID_FUNCTIONAL_RANGE to 50
- PID_K1 smoothing to 0.5

### Steps to Reproduce

Flash the device to the firmware as described and try to print a calibration cube at 205C.

**Expected behavior:**

Stabilizing temperatures and printing.

**Actual behavior:**

Temperature never stabilizes. Device is stuck in the heating phase.

EvilGremlin, 2023-12-24:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* 3D Printing Discord server. Join link: https://discord.gg/3dprinters
* Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

thinkyhead, 2024-01-10:
PID tuning can sometimes be tricky. You now also have the option to use MPC, so give that a try. As mentioned above, you should seek help in the help areas. With interactive help from the community it's very likely you will get your problem solved pretty quickly.

closed 2024-01-10 by thinkyhead