MarlinFirmware/Configurations #999: CTC Bizer Marlin Not Compiling Arduino

MechEng2016, 2024-01-02:
I used an Arduino compiler and set it for Arduino Mega 2560 board.  My CTC Bizer uses a 2560 ATmega chip.  It appears to give similar errors between the Marlin 2.0, 2.1 and the bug fixes.  I am am trying to make a hex file to directly upload to the CTC board.  I was able to upload a Marlin hex file another person created. The hex file worked but, I wanted to make some changes.  Below is about 1/8 of the errors that I received.  Anyone know what the main issue is and how to correct it?  I can provide more of the error messages as needed.

In file included from C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\HAL\AVR\HAL.h:30:0,
                 from C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\HAL\HAL.h:30,
                 from C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\inc\MarlinConfig.h:33,
                 from C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\lcd\marlinui.h:24,
                 from C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\lcd\HD44780\lcdprint_hd44780.cpp:41:
C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\pins\mega\pins_MIGHTYBOARD_REVE.h:227:51: error: pasting "/* G5*/" and "_RPORT" does not give a valid preprocessing token
     #define BEEPER_PIN                         4  // G5
                                                   ^
C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\HAL\AVR\fastio.h:84:39: note: in definition of macro '_TOGGLE'
 #define _TOGGLE(IO)           (DIO ## IO ## _RPORT = _BV(DIO ## IO ## _PIN))
                                       ^~
C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\libs\buzzer.h:64:43: note: in expansion of macro 'TOGGLE'
       FORCE_INLINE static void invert() { TOGGLE(BEEPER_PIN); }
                                           ^~~~~~
C:\Users\****\Documents\Marlin-bugfix-2.1.x\Marlin-bugfix-2.1.x\Marlin\src\libs\buzzer.h:64:50: note: in expansion of macro 'BEEPER_PIN'
       FORCE_INLINE static void invert() { TOGGLE(BEEPER_PIN); }

ellensp, 2024-01-02:
BOARD_MIGHTYBOARD_REVE is not a mega 2560 board.
It has additional pins  that are not defined with selecting Arduino Mega 2560

closed 2024-01-02 by ellensp

ellensp, 2024-01-02:
Use vscode, it will install correct board file for BOARD_MIGHTYBOARD_REVE