MarlinFirmware/Configurations #1000: BIBO extruder PID parameters that work (for my printer)

psehorne, 2024-01-05:
### Bug Description

This is not exactly a bug, just an  suggested improvement for the BIBO 2 Touch extruder PID parameters. 

The default extruder PID parameters allow my BIBO extruders to vary +/- 8C.  Auto tuning PID did no better.  
I found the following PID parameters much better.  My extruders now settle in within +/- 1C.

In configuration.h
    #define  DEFAULT_Kp 8
    #define  DEFAULT_Ki  2
    #define  DEFAULT_Kd 200

[BIBO_configuration_files.zip](https://github.com/MarlinFirmware/Configurations/files/13844866/BIBO_configuration_files.zip)

### Configuration Files

psehorne, 2024-01-05:
With target temp set to 215C, and ...

Set PID M301 E0 P23.89 I1.37 D104.5.  Note power varying from 0% to 100%
[image]

Set PID M301 E0 P8 I2 D200.  Note power hovering consistently near 50%.
[image]

thinkyhead, 2024-01-10:
As you've discovered, PID parameters that work with one particular machine build may not work with other builds, as every machine varies in its construction. One hotend may use hardened steel block with a cemented heater while another may have an aluminum block with a screwed-in heater, and all components down to the choice of braided copper or unbraided steel wires, which will have slightly different reluctances, will vary according to price and availability at the time of machine construction.

Therefore, every machine must be individually tuned, and you also have the option to use MPC instead of PID. I'm happy to apply your PID values to the examples, but there is no way to know whether your machine build is typical or if it is an outlier without getting more testers.

closed 2024-01-10 by thinkyhead