MarlinFirmware/Configurations #1001: [BUG] Wrong Pin documentation in MKS Sbase README.md

FriedSemiconductors, 2024-01-06:
### Description:
In the [README.md](https://github.com/MarlinFirmware/Configurations/blob/release-2.1.2.1/config/examples/Mks/Sbase/README.md) of the **MKS Sbase** example are notes about swapping pins of the **EXP2** connector to get the **RepRapDiscount Full Graphic Smart Controller** to work.
One of the mentioned pins is the pin `P0.5`, the pin is not part of **EXP2** as mentioned, but instead a pin of the **X-Motor** driver.

Does another Pin have to be swapped or can we ignore this line? And what are the corresponding target pins on `J8`, they have a different style of labling (`GND`; `P1.22`; `P1.23`; `P2.12`; `P2.11`; `P4.28` > see screenshot).

Can someone please clarify, as i am unsure on how to wire it correctly with this information?

I'd really appreciate the help, thank you for your time and have a nice day.

### Screenshots:

[image]

[image]

[image]

thisiskeithb, 2024-01-06:
It should probably be P0.9.

https://github.com/MarlinFirmware/Marlin/blob/2a8c00bdeb5268b11a3bbddfc8797a3f09a92947/Marlin/src/pins/lpc1768/pins_MKS_SBASE.h#L172-L190:

```c
  /**
   * A custom cable is needed. See the README file in the
   * Marlin\src\config\examples\Mks\Sbase directory
   * P0.27 is on EXP2 and the on-board SD card's socket. That means it can't be
   * used as the SD_DETECT for the LCD's SD card.
   *
   * The best solution is to use the custom cable to connect the LCD's SD_DETECT
   * to a pin NOT on EXP2.
   *
   * If you can't find a pin to use for the LCD's SD_DETECT then comment out
   * SD_DETECT_PIN entirely and remove that wire from the the custom cable.
   */
  #define SD_DETECT_PIN                    P2_11  // J8-5 (moved from EXP2 P0.27)
  #define SD_SCK_PIN                       P1_22  // J8-2 (moved from EXP2 P0.7)
  #define SD_MISO_PIN                      P1_23  // J8-3 (moved from EXP2 P0.8)
  #define SD_MOSI_PIN                      P2_12  // J8-4 (moved from EXP2 P0.9)
  #define SD_SS_PIN                        P0_28
  #define SOFTWARE_SPI                            // With a custom cable we need software SPI because the
                                                  // selected pins are not on a hardware SPI controller
```

Also, these MKS examples shouldn't even really exist in this repo since they're just a generic board configs and not for a real printer.

closed 2024-01-10 by thinkyhead

FriedSemiconductors, 2024-01-12:
@thisiskeithb 
thank you for clearing this up, that helps a lot.

commit f2a25dc1 (2024-01-10): 📝 MKS SBASE custom cable correction

Fixes #1001